ARM-software/CMSIS_5 #558: RTX5 - Is it able to dynamic calculate OS_DYNAMIC_MEM_SIZE?I

kangjianbin, 2019-03-24:
I have a LPC54606 board and use its SRAMX memory as RTX5's memory. 0x040000000 ~ 0x04008000.
The memory map is something like:
```
--------
(.bss.os*  .data.os *) (OS_DYNAMIC_MEM_SIZE is defined as 0, so os_mem consumes 0 bytes)
--------
Heap  (Will used it as global dynamic memory)
-------
Main Stack (1KB)
------
```

I can calculate Heap's address and size via linker's exported symbol. However looks like it is impossible to modify global-dynamic-memory size in run time as osRtxConfig is declared as constant.

JonatanAntoni, 2019-03-25:
Hi @kangjianbin,

Please be aware that the global-dynamic-memory of RTX is not the same than the default C/C++ heap. RTX global dynamic memory is only used to allocate RTOS objects. RTX5 assumes the user knows the memory requirements of the application at compile time. This is widely accepted for microcontroller and real-time applications.

Do you simply want to use all free memory as RTX5 global dynamic memory? Is it feasible for you to update RtxConfig.h accordingly? I.e. after doing a dry compilation run?

Cheers,
Jonatan

kangjianbin, 2019-03-25:
Hi @JonatanAntoni ,

Thanks for your answer. Yes, I want to use all free memory in SRAMX as global dynamic memory.
Looks like I can use the follow step to detect available free memory:
1. Set OS_DYNAMIC_MEM_SIZE to 0
2. Compile the executable. Then we can get available memory: Total - (.bss.os* + .data.os*) - STACK_SIZE
3. Set OS_DYNAMIC_MEM_SIZE to available memory and recompile to check if the size is correct.

But is there any way to modify OS_DYNAMIC__MEM_SIZE in run-time? 
I can calculate size of free memory via linker's export symbols.

Thanks,
Jianbin

JonatanAntoni, 2019-03-25:
Hi @kangjianbin,

I guess I got your point. Unfortunately you cannot change the size at run time because all memory is statically allocated (static array in rtx_lib.c). This way its not even (easily) possible automatically resize the memory by using the linker output.

Cheers,
Jonatan

kangjianbin, 2019-03-25:
@JonatanAntoni Thanks!

closed 2019-03-25 by kangjianbin