ARM-software/CMSIS_5 #561: SVDConv: __SAU_REGION_PRESENT value is always 1

tu44okawa3, 2019-04-02:
## Problem
When generating SVD file without sauNumRegions and sauRegionsConfig elements, __SAU_REAGION_PRESENT value was generated as 1.

According to the SVD description format, the occurrences of sauNumRegions and sauRegionsConfig are 0..1. So My understanding is that if there are not those elements, the value of __SAU_REGION_PRESENT should be generated as 0.

### Error result
`#define __SAU_REGION_PRESENT           1        /*!< SAU present`

### Expected results
`#define __SAU_REGION_PRESENT           0        /*!< SAU present`

## Reproducible svd file
SVD file which commented out sauNumRegions and sauRegionsConfig elements from the following file:
(ARM)\CMSIS\5.4.0\Device\ARM\SVD\ARMCM23.svd

[ARMCM23_NoSAU.svd.txt](https://github.com/ARM-software/CMSIS_5/files/3033503/ARMCM23_NoSAU.svd.txt)

## Environment information
- OS: Windows 10
- SVDConv version: V3.3.25
- Execute command : SVDconv.exe ARMCM23_NoSAU.svd --generate=heade

JonatanAntoni, 2019-04-02:
Hi @tu44okawa3,

That seems to be an non-intuitive default used in SVDConv when `sauNumRegions` is not given. As a workaround you can add `<sauNumRegions>0</sauNumRegions>` to clearly state that no SAU is available at all.

We will revise the default values used.

Cheers,
Jonatan

thorstendb-ARM, 2019-04-05:
Hi @tu44okawa3,

I've fixed the issue and created a new version. Can you please test?

BR,
Th. de Buhr

tu44okawa3, 2019-04-07:
Hello @thorstendb-ARM and @JonatanAntoni ,

I confirmed that it was fixed.
Thank you so much for fixing it so quickly.

Best regards,
tu44okawa

closed 2019-04-07 by tu44okawa3

thorstendb-ARM, 2019-04-08:
Hi @tu44okawa3,

Thank you for testing!

BR,
/th.