ARM-software/CMSIS_5 #562: Math functions arm_sin_f32 & arm_cos_f32 functions always return zero

broccolee90210, 2019-04-02:
Hi,

I am using keil uVision V5.25.2.0, MQX, and kinetis K64f. I can't use arm_sin_f32 & arm_cos_f32. They always return zero. I have tested a few other functions like arm_add_f32 and it works. I haven't tested all the functions in the DSP library to see if they all work. 

How can I troubleshoot this?

Thanks.

JonatanAntoni, 2019-04-03:
Hi @broccolee90210,

have you recognized the input value range for sin and cos functions? See http://arm-software.github.io/CMSIS_5/DSP/html/group__sin.html#gae164899c4a3fc0e946dc5d55555fe541

The source code of DSP Library is freely available so you can debug into the implementation yourself. Or consider providing us a reproducer, i.e. how do you call the function, which result do you expect and which result do you actually get?

Cheers,
Jonatan

broccolee90210, 2019-04-03:
Hi @JonatanAntoni,

Thanks for the suggestion! I am able to use it properly when I copied the source code and the sinTable_f32 Look Up Table into my code. Hmmmm, this could mean that some of the source codes & libraries are not included properly via Keil Manage Runtime Environment... I tried including the CMSIS DSP library the same way on a different system and got the same result.

Thanks.

JonatanAntoni, 2019-04-03:
Hi @broccolee90210,

Using Keil MDK and RTE Manager you should get the very same results when switching force and back between library and source code variant of CMSIS DSP component. The latter variant should be available given you are using the latest CMSIS Pack 5.5.1. 

Its possible that anything got wrong during building the libraries. May I ask you to clearly verify that you get different results using CMSIS Pack 5.5.1 and CMSIS DSP 1.6.0 when switching between library and source variants, please?

Cheers,
Jonatan

christophe0606, 2019-05-29:
It looks like it is not CMSIS-DSP related but more of a problem with building with Keil.

@broccolee90210 Have you been able to solve your problem ?

closed 2019-12-17 by christophe0606