ARM-software/CMSIS_5 #563: Difficulty using convolve nonsquare

atrevely, 2019-04-02:
I'm having difficulty troubleshooting the output from the `arm_convolve_HWC_q7_basic_nonsquare` layer. For testing, I am trying to input 4x1 data in two channels, like so:

`#define INPUT_DATA {1, 2, 3, 4, 5, 6, 7, 8}
#define CONV1_IN_X 4
#define CONV1_IN_Y 1
#define CONV1_OUT_X 4
#define CONV1_OUT_Y 1
#define CONV1_IN_CH 2
#define CONV1_OUT_CH 1`

I also am simply sliding a single "box" over, so kernel dimension is 1 for x and y. Stride is 1 (although there is no striding to be done in the y dimension, I'm unsure if this matters).  No padding or shifts. So we have:

`#define CONV1_KERN_X 1
#define CONV1_KERN_Y 1
#define CONV1_PAD_X 0
#define CONV1_PAD_Y 0
#define CONV1_STRIDE_X 1
#define CONV1_STRIDE_Y 1
#define CONV1_OUT_SHIFT 0
#define CONV1_BIAS_SHIFT 0`

The weights and biases are simply:

`#define CONV1_WEIGHTS {1, 2}
#define CONV1_BIASES 0`

I run this using the following function:

`arm_convolve_HWC_q7_basic_nonsquare(input_data, CONV1_IN_X, CONV1_IN_Y, CONV1_IN_CH, conv1_wt, CONV1_OUT_CH, CONV1_KERN_X, CONV1_KERN_Y, CONV1_PAD_X, CONV1_PAD_Y, CONV1_STRIDE_X, CONV1_STRIDE_Y, conv1_bias, CONV1_BIAS_SHIFT, CONV1_OUT_SHIFT, data_buffer1, CONV1_OUT_X, CONV1_OUT_Y, (q15_t *) conv_buffer, NULL);`

And I use the following buffers:

`q7_t *data_buffer1[4];
q7_t conv_buffer[2 * 2 * 4];`

As I step through the code, I get an error in data_buffer1:  `error: Cannot access memory at address`

Can anyone spot anything obvious that I'm doing wrong? I would expect an output vector with four entries, something like [11, 14, 17, 20], although I'm unsure on ordering (part of the reason for testing) so it may be [5, 11, 17, 23].

Any help would be greatly appreciated.

majianjia, 2019-04-03:
Hi @atrevely 

I think it is only a simple mistake, you defined your output buffer as `q7_t *data_buffer1[4]`. However, it should be `q7_t data_buffer1[4]`.

atrevely, 2019-04-04:
Awesome! Thanks. That now allows it to produce an output from the layer, however the output is not what I was expecting. For example:

Passing a 4x1 input data vector in 2 channels as follows:

`#define INPUT_DATA {1, 2, 3, 4, 5, 6, 7, 8}`

I get an output of [65, 71, 77, 83]. Sine my weights are [1, 2] and bias is 0, these numbers are far too large (see my expected output from original post). Any idea how these numbers are being calculated? Is there a shift that needs to happen?

atrevely, 2019-04-04:
Another update after some testing. The value set for the bias seems to be having an unexpected impact on the output.

For `#define CONV1_BIASES 1` the output is [4, 10, 16, 22], which would be the correct output for a bias of -1.

For `#define CONV1_BIASES 2` the output is [5, 11, 17, 23], which is the correct output for bias 0.

For `#define CONV1_BIASES 2` the output is [37, 43, 49, 55], and now I'm lost.

majianjia, 2019-04-05:
I agree with your original post. 
[5, 11, 17, 23] should be the results of bias=0. 

Please check if your data (bias) has been set in the right way. 

this is what you should probably do. 
`q7_t conv1_bias[] = {CONV1_BIASES};`  then you can pass it to conv2d as `conv1_bias`
`q7_t conv1_bias = CONV1_BIASES;`  then you can pass it to conv2d as `&conv1_bias`
if you are not writing in a similar way, the reason is the same as what I have explained in my previous post. the misuse of the pointer and value. 

A sign is your bias setting, with 2 and 3
[37, 43, 49, 55] - [37, 43, 49, 55] = [32, 32, 32, 32], where inidcated that you have set bias to 32 instead of 3.

atrevely, 2019-04-08:
Perfect, that was the issue. I don't have much experience in C (obviously), so setting these variables is tricky, but I think it came down to not having the [] after the bias assignment when there is only one bias. Thanks for all of your help.

Now, of course, I realize that ARM doesn't have any nonsquare pooling functions...

majianjia, 2019-04-09:
@atrevely 
It is, pointers in C, especially with an array, is tricky. 

For nonsquare pooling...
The good news is they do have one but hidden in another project. 
[Average Pooling on KWS](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Source/local_NN/arm_avepool_q7_HWC_nonsquare.c)
The bad news is it is not optimised. 

I have similar implementations for both max pooling and avg pooling. 
[NNoM local backend](https://github.com/majianjia/nnom/blob/master/src/nnom_local.c)

closed 2019-06-05 by felix-johnny