ARM-software/CMSIS_5 #567: Numpy compatible Python wrapper for CMSIS-DSP

christophe0606, 2019-04-11:
Dear CMSIS-DSP users,

I have recently pushed an experimental Python wrapper for CMSIS-DSP which is compatible with Numpy and which can be used on any laptop/desktop.

You can build your signal processing flow with scipy then progressively replace some part of it with CMSIS-DSP and experiment with saturation, rounding effects until you have something ready for a board.

More details are in the README in folder DSP/PythonWrapper. I am not a Python expert and the installation procedure can be improved.

Also it is a very first release and I have not tested all the functions in the wrapper although the main ones (like biquad, fir, FFT …) were tested.

I hope it will be useful and I am waiting for your feedback.

maugar, 2019-09-27:
In the README, a custom `arm_math.h` is mentioned.
I was interested in reading the source code to see if a similar approach can be used to integrate CMSIS-DSP in a ceedling-based testing framework, but it looks like it's missing from [CMSIS/DSP/PythonWrapper/cmsisdsp_pkg/src](https://github.com/ARM-software/CMSIS_5/tree/master/CMSIS/DSP/PythonWrapper/cmsisdsp_pkg/src) folder.

christophe0606, 2019-09-27:
@maugar This custom arm_math.h is no more needed. (I'll update the README).

Now you can use the standard one. If you compile with MSVC it will work. It you use another compiler you may define __GNUC_PYTHON__ 

This is needed to redefine following macros and select C implementation of some intrinsincs like __SSAT

```
__ALIGNED(x) 
__STATIC_FORCEINLINE 
__STATIC_INLINE
LOW_OPTIMIZATION_ENTER
LOW_OPTIMIZATION_EXIT
IAR_ONLY_LOW_OPTIMIZATION_ENTER 
IAR_ONLY_LOW_OPTIMIZATION_EXIT
```

maugar, 2019-09-27:
> you may define `GNUC_PYTHON`

This information was very useful, thanks 🙏 

It's worth sharing that, as of today, support for `GNUC_PYTHON` define is implemented in develop branch and I was using the master branch.

christophe0606, 2019-09-30:
@maugar Right ! It is a recent change and not yet in the main branch.

LukeGary462, 2021-02-12:
is there any chance that this will make it on PyPi ?

christophe0606, 2021-02-15:
@LukeGary462 At some point yes. But I need to improve the setup.py script.

closed 2021-09-30 by christophe0606