ARM-software/CMSIS_5 #568: SVD example for Cortex-M33

radhika-raghavendran, 2019-04-11:
The latest distribution of CMSIS pack v5.5.1 does not contain the proper SVD example and template for Cortex-33 devices with TrustZone and FPU.
The example SVD does not match the description in https://arm-software.github.io/CMSIS_5/SVD/html/elem_cpu.html 
This description contains fields for configuration of SAU and generation of the partition header file. 
Can you tell me where I can find the latest example for SVD matching with Version 1.3.3 of the SVD Schema?

Thank you.

JonatanAntoni, 2019-04-11:
Hi @radhika-raghavendran,

I guess you are referring to this example?

https://github.com/ARM-software/CMSIS_5/blob/23f36583da1ce548c8457c368fdd610e72efb073/Device/ARM/SVD/ARMCM33.svd#L1-L103

What exactly are you missing? Or what do you think is wrong with that example?

Cheers,
Jonatan

radhika-raghavendran, 2019-04-11:
Hi Jonatan,

Thanks for the quick response. I was looking at > CMSIS\SVD  folder in the ARM CMSIS installation on my system. I should have checked > Device\ARM\SVD  folder instead.

radhika-raghavendran, 2019-04-11:
Hi Jonatan,
I have one more query with respect to SVD. Can we skip some interrupts in the <interrupt> tag under the <peripheral> tag?
What I want to do is something like this, right after cpu information in SVD. When I declare something like this I get an error from SVDConv.exe **"External Interrupts must start on position 0"** Can you please help confirm the SVDConv behaviour?
`<peripherals>
 
    <peripheral>
      <name>I2C0</name>
      <version>1.0</version>
      <description>I2C peripheral</description>
      <groupName>I2C</groupName>
      <baseAddress>0x4003C000</baseAddress>
      <size>32</size>
      <access>read-write</access>

      <addressBlock>
        <offset>0</offset>
        <size>0x1C</size>
        <usage>registers</usage>
      </addressBlock>

      <interrupt>
        <name>I2C0</name>
        <description>I2C0 interrupt</description>
        <value>209</value>
      </interrupt>

      <registers>
</peripheral>
  </peripherals>
</device>`

jkrech, 2019-04-11:
Can you please attach the complete SVD file that creates this error message?

radhika-raghavendran, 2019-04-11:
I am invoking CMSIS-SVD SVD Consistency Checker / Header File Generator V3.3.25 which comes with CMSIS 5.5.1.
Attached is the SVD file, command script, error log for a new Cortex-M33 device we are trying to bring up.

Thanks.
[SVD.zip](https://github.com/ARM-software/CMSIS_5/files/3068241/SVD.zip)

jkrech, 2019-04-11:
Thanks! 
Reviewing your example it became clear that you are in the early stages and this Error should much rather be a Warning. This check is assuming that all devices will at least use one of the external interrupts 0 - 15.
There is two ways of counting external interrupts. You either start with 0 (this is how it is done within CMSIS) or you start counting external interrupts with 16 (continue from counting the core interrupts 0 - 15).  This check shall ensure that the CMSIS counting is used.

For the time being you could just add a dummy interrupt to your peripheral:
      <interrupt>
        <name>dummy</name>
        <description>Dummy Interrupt</description>
        <value>0</value>
      </interrupt>

We will investigate how to best address this in SVDConv going forward.
Sorry for the confusion.

radhika-raghavendran, 2019-04-11:
Thanks for the clarification. I went ahead and added a dummy interrupt for external interrupt 0  to the SVD. The same error persists for some reason. Looks like I need to add all interrupts till the interrupt for I2C. The only problem is that this interrupt in question is at 210 and I need to add 209 interrupts as dummy before this :)

jkrech, 2019-04-11:
Well, I tried this using your svd file and your command line but I am not getting an error anymore.

radhika-raghavendran, 2019-04-11:
Thanks. It works now. Just added the interrupt tag in I2C peripheral and it generated the header file,

closed 2019-04-11 by radhika-raghavendran