ARM-software/CMSIS_5 #569: How to print some parameter?

ccu1tn, 2019-04-15:
Dear everyone;

In this picture, there are some value like  " Im_in" ," dim_im_in"
[image] 
So. How to print the value Im_in and dim_in_in by c++?
Many thank everyone share

majianjia, 2019-04-15:
hi, 
I am not entirely sure the "print" in your term.
you should already have these parameters before you can use this function.

Im_in is the pointer to the input image.
 Dim_im_in is the dimension of the input image, the image is default to square shap, so you only need to pass once.

If you want to show image, you might consider other lib. For example, opencv can show an image.

ccu1tn, 2019-04-16:
Hi @majianjia ;
Thank you so much.

closed 2019-04-16 by ccu1tn