ARM-software/CMSIS_5 #573: DSP lib linker syntax error

dizcza, 2019-04-17:
Hello, 

`libarm_cortexM4lf_math.a` is not linked correctly since https://github.com/ARM-software/CMSIS_5/commit/41d9e3577cd61c0380f5c1fa5f3008dd5803f30e

I changed the path accordingly.

The previous version of `libarm_cortexM4lf_math` works fine.

```
/opt/Atollic_TrueSTUDIO_for_STM32_x86_64_9.3.0/ARMTools/bin/../lib/gcc/arm-atollic-eabi/6.3.1/../../../../arm-atollic-eabi/bin/ld:/home/dizcza/microcontrollers/CMSIS_5/CMSIS/DSP/Lib/GCC/libarm_cortexM4lf_math.a: file format not recognized; treating as linker script
/opt/Atollic_TrueSTUDIO_for_STM32_x86_64_9.3.0/ARMTools/bin/../lib/gcc/arm-atollic-eabi/6.3.1/../../../../arm-atollic-eabi/bin/ld:/home/dizcza/microcontrollers/CMSIS_5/CMSIS/DSP/Lib/GCC/libarm_cortexM4lf_math.a:1: syntax error
```

JonatanAntoni, 2019-04-23:
HI @dizcza,

can you check the file size on your disk, please?

We started to use Git-LFS for binary files a while ago. This is to limit the repo growth, e.g. by adding large binary which in turn needs to be cloned. I suspect you need to install Git-LFS properly on your system and fetch the actual libraries from binary storage.

Apologies for any inconvenience,
Jonatan

dizcza, 2019-04-25:
Indeed, I haven't noticed updates in readme.
With git lfs it builds fine.

closed 2019-04-25 by dizcza