ARM-software/CMSIS_5 #574: osMutexRelease issue

eugene345, 2019-04-19:
Hi,

I'm debugging an issue, where osMutexRelease doesn't release mutex correctly in some cases (at least it looks so).

This is the first problem I found already:

```
osMutexId_t m = osMutexNew(NULL);
osMutexAcquire(m, 0);
osMutexRelease(m);
```

After this code execution the mutex's control block member "owner_thread" still points to the ex-owner. I think, it should be NULL.

Best regards,
Eugene

RobertRostohar, 2019-04-19:
"lock" member of the mutex control block is used to determine if the mutex is acquired `lock != 0` or not `lock == 0`. When the mutex is released and `lock == 0` the value of "owner_thread" is ignored.
Therefore it is not necessary that the value of "owner_thread" should be NULL. This is expected behavior.

eugene345, 2019-04-19:
Yes, seems "owner_thread" used only in the component viewer window for RTX:

[image]

Maybe still makes sense to set it to NULL to make information more clear in this window?

RobertRostohar, 2019-04-19:
This can be handled in the RTX component viewer without setting it to NULL.

eugene345, 2019-04-19:
Yes, but it is not handled there.

RobertRostohar, 2019-04-19:
Currently it is not handled but can be without the need to change RTX code.

eugene345, 2019-04-19:
I've added a software breakpoint here: 

```
/// Acquire a Mutex or timeout if it is locked.
/// \note API identical to osMutexAcquire
static osStatus_t svcRtxMutexAcquire (osMutexId_t mutex_id, uint32_t timeout) {
  os_mutex_t  *mutex = osRtxMutexId(mutex_id);
  os_thread_t *thread;
  osStatus_t   status;

  // Check running thread
  thread = osRtxThreadGetRunning();
  if (thread == NULL) {
    EvrRtxMutexError(mutex, osRtxErrorKernelNotRunning);
    //lint -e{904} "Return statement before end of function" [MISRA Note 1]
    return osError;
  }

  // Check parameters
  if ((mutex == NULL) || (mutex->id != osRtxIdMutex)) {
    EvrRtxMutexError(mutex, (int32_t)osErrorParameter);
    //lint -e{904} "Return statement before end of function" [MISRA Note 1]
    return osErrorParameter;
  }

  // Check if Mutex is not locked
  if (mutex->lock == 0U) {
    // Acquire Mutex
    mutex->owner_thread = thread;
    if(mutex == thread->mutex_list) __BKPT(0); // TEST CODE!!!!!!!!
    mutex->owner_next   = thread->mutex_list;
    mutex->owner_prev   = NULL;
```
it should never stop there, right?

RobertRostohar, 2019-04-19:
Correct. You might have some memory corruption (stack overlofw, ...) which could eventually lead to this.

eugene345, 2019-04-21:
Seems, thread->mutex_list isn't updated if mutex->owner_prev != NULL and it keeps pointing to the released mutex:

https://github.com/ARM-software/CMSIS_5/blob/837ed048639cb9296e1a96a5e10170d83972d8ee/CMSIS/RTOS2/RTX/Source/rtx_mutex.c#L291-L299

I guess, this could be an issue?

JonatanAntoni, 2019-04-23:
Hi @eugene345,

As far as I got your feedback correct you are pointing out some "optimizations" that are misleading. Pointers are not (always) reset to `NULL` if they become irrelevant. This is not an issue per-se, but might be misleading when shown in the RTOS window of an IDE (or when stepping through the code). The decision to be made is between robustness and efficiency. Currently its implemented with efficiency over robustness.

I understand that we should think about enhancing the RTX5 component viewer to "hide" the owner thread if the Mutex is not locked. I assess this as a valuable improvement.

The second analysis you did is not yet fully clear to me. As Robi already pointed out these situations usually occur when some memory corruption happened before. Are you able to reproduce this issue? Is it feasible for you to share the reproducer?

Thanks for your feedback,
Jonatan

eugene345, 2019-04-23:
Hi @JonatanAntoni,

My custom system is quite complicated and uses many technologies (W-Fi, eMMC, etc). A problem with mutexes happens sporadically and not often. I can't find regularity and because of that I can't create a reproducer for you. 

I don't think, it is a memory corruption. Stack overrun checking and stack usage watermark enabled. And I can see that stack overflow doesn't happen and stack has enough reserve.

I've started to analyse the RTX code and found the issue (in my view) I've explained in my latest comment. 
So, in some cases (if mutex->owner_prev != NULL) thread's mutex_list isn't updated and points to the released mutex. I think, this is a serious problem, which harms the correct mutex operation.
Could you check, please, is it true or did I miss something?

Thanks,
Eugene

JonatanAntoni, 2019-04-23:
Hi @eugene345,

Okay, I got your issue. Sporadic issues are always hard to analyze. Reasonable that you can't provide a reproducer. Bad luck.

The bad thing is something goes completely wrong in your application. What you are observing in the debugger is only the consequence of the root cause. A more robust (and less optimized) implementation might hide the underlying issue. I think we should try to understand whats actually going wrong.

I'll try to figure out possible causes and report back.

Cheers,
Jonatan

JonatanAntoni, 2019-04-23:
Hi @eugene345,

It's very hard to predict where the error comes from. If you are sure its not due to memory corruption, e.g. due to stack overflow, it might be a priority issue of the RTOS interrupts (SVC, SysTick, PendSV). Please double check these as well. Can you provide us some details about the device and other components you are using?

In fact the situation you are describing above, i.e. `thread->mutex_list == mutex` but `mutex->owner_prev != NULL`, is already broken. `mutex_list` is the pointer to head of the linked list chaining all Mutexes owned by a certain Thread. The `owner_prev` should always be `NULL` for the first Mutex in that list. If this is not the case anymore something got screwed up. We need to figure out when that happens.

You could try to use Event Recorder to get a better idea of whats going on in your actual system. You might want to enable only Mutex events with levels Error, API and Op, but no Details. If its feasible for you please share the event log.

Please consider involving the commercial customer support as well. The colleagues have probably more possibilities to assist you than what we can do in public here on GitHub.

Cheers,
Jonatan

eugene345, 2019-04-23:
Hi @JonatanAntoni,

I think, I found the problem finally. Please, check my pull request #578

Best regards,
Eugene

JonatanAntoni, 2019-04-24:
Hi @eugene345,

Thanks for digging that deep into the details. It seems you finally figured out the concrete situation where it fails for you. With that information at hand is it feasible for you to create a reproducer we could use as a regression test? Or can you sketch e.g. a sequence diagram to facilitate the comprehensibility?

Cheers,
Jonatan

eugene345, 2019-04-24:
Hi @JonatanAntoni,

Here is a reproducer:

```C
osThreadId_t t1, t2;  
osMutexId_t m1, m2;

void thread1(void *argument)
{
  osMutexAcquire(m1, 0);
  osDelay(5);
  osMutexRelease(m1);
}

void thread2(void *argument)
{
  osStatus_t stat;
	
  stat = osMutexAcquire(m2, 0);
  stat = osMutexAcquire(m1, 10);
  // At this point linked list is broken (owner_prev of m2 doesn't point to m1)
}

void test(void) 
{  
  m1 = osMutexNew(NULL);
  m2 = osMutexNew(NULL);
  t1 = osThreadNew(thread1, NULL, NULL);
  t2 = osThreadNew(thread2, NULL, NULL);
}
```
Cheers,
Eugene

JonatanAntoni, 2019-04-24:
Hi @eugene345,

Great! Looks fairly easy to reproduce.
Thank you very much for your contribution, its highly appreciated.

Cheers,
Jonatan

JonatanAntoni, 2019-05-21:
Hi @eugene345,

May I ask you to close this issue if the provided fix solves your concerns, please?

Thanks,
Jonatan

eugene345, 2019-05-21:
Hi @JonatanAntoni,

This bug is fixed.

Cheers,
Eugene

closed 2019-05-21 by eugene345