ARM-software/CMSIS_5 #575: A suggestions

lswzjuer, 2019-04-23:
Hope to be able to join the  support of binarization convolution which is  based on Xnor and bitcountering operations

JonatanAntoni, 2019-04-23:
Hi @lswzjuer,

Can you update this issue to be slightly more expressive, please?
What exactly do you want to suggest?
-> Update the title accordingly, should be the rough summary instead of just "A suggestion".
-> Rephrase the description to clearly what you want to achieve, why, and how if feasible.

Thanks for getting in touch,
Jonatan

majianjia, 2019-04-23:
I think op is referencing this work: https://pjreddie.com/media/files/papers/xnor.pdf

lswzjuer, 2019-04-23:
> Hi @lswzjuer,
> Can you update this issue to be slightly more expressive, please?
> What exactly do you want to suggest?
> -> Update the title accordingly, should be the rough summary instead of just "A [...]
We can binarize the activation values and weights to constrain them to {-1, +1}，If all of the operands of the convolutions are binary, then the convolutions can be estimated by XNOR and bitcounting operations without Multiplication and addition. 
XorNet has so much potential, I suggest you in the Source/ConvolutionFunctions provides support for binary convolution (just like arm_convolve_HWC_q2. c)

JonatanAntoni, 2019-04-23:
I tend to remove this issue because it's still not comprehensible for me what the request is about.

On the first glance it sounds like some enhancement to the CMSIS-DSP library. But reading "activation values" makes me think you are talking about CMSIS-NN functionality. Please bare in mind not everybody reading here is an expert in DSP or NN. CMSIS covers a broad range of aspects. I doubt "a suggestion" tells anything to a wider audience.

Please consider creating a new issue, stating clearly what you want to achieve, why and preferable how.

closed 2019-04-23 by JonatanAntoni