ARM-software/CMSIS_5 #576: Should NVIC_SetVector have a DSB?

kjbracey-arm, 2019-04-23:
I saw some code for Cortex-M7 where someone had done:

        NVIC_SetVector(irq_n, vector);
    #if defined (__ICACHE_PRESENT) && (__ICACHE_PRESENT == 1U)
        SCB_InvalidateICache();
    #endif
        NVIC_EnableIRQ(irq_n);

I queried the invalidate, and they said they had seen a crash without it.

Now, I believe their actual issue is that you do architecturally need a `DSB` between setting a vector and enabling it, although ARM App Note 321 says M0, M0+, M1, M3 and M4 can manage without.

I think we should do one of the following, but I can't decide which:

* Add `DSB` to all `NVIC_SetVector`
* Add `DSB` to all `NVIC_SetVector` except M0,M0+,M1,M3,M4 (because we believe App Note 321)
* Add `DSB` to all `NVIC_EnableIRQ`
* Add `DSB` to all `NVIC_EnableIRQ` except M0,M0+,M1,M3,M4
* Leave it and document it with a note in each call.

(I do also have #493 open suggesting adding a compiler barrier to `NVIC_EnableIRQ` - putting a `DSB` there would supersede that. I think I favour putting it in `NVIC_SetVector` or nowhere anyway).

JonatanAntoni, 2019-04-23:
Hi @kjbracey-arm,

Thanks for pointing this out. I guess it only affects Cortex-M7, and probably Cortex-A?

I think we should be minimalistic, e.g. add this only to the affected targets. And the `DSB` seems to belong to the `NVIC_SetVector` rather than to `NVIC_EnableIRQ`, i.e. we need to assure that the IRQ handler pointer has been completely before enabling the interrupt. Additionally, I'd expect `NVIC_SetVector` to be used less often than `NVIC_EnableIRQ`. Hence the introduced overhead would be smaller if adding it to `NVIC_SetVector`.

I don't understand how #493 is involved here. I expect the `DSB` would be needed *before* enabling the IRQ. But in #493 I understand we might need an additional `DSB` after enabling the interrupt? *puzzled*

Cheers,
Jonatan

kjbracey-arm, 2019-04-23:
Agree that it feels like it should be in `SetVector`.

Cortex-A (AArch32) has a totally different exception vector system, so this doesn't apply, at least to the in-tree GIC code.

I've not found any public info about later M cores - ARM now seems more reluctant to publish "implementation details" like App Note 321. I've heard this apparent evidence that the M7 does fail, which is unsurprising given its fancy store buffer.

If I was guessing, I'd reckon the M23 and M33 wouldn't need a DSB, but I've been wrong before about assuming the M33 is "just an ARMv8-M M4".  I'd be inclined to add a DSB to everything that wasn't explicitly documented in App Note 321.

(#493 was mainly suggesting putting a compiler barrier **before** enabling the interrupt to ensure that all accesses made within a block protected by `DisableIRQ`/`EnableIRQ` stayed inside it.  Any DSB put in for the vector change reason would also act as that barrier.  My example there was pondering a second barrier after the NVIC enable store, but that was possible overcaution. I was comparing with atomics where "store release" semantics need a barrier before the store, and "sequential consistency" requires an extra barrier after.)

kjbracey-arm, 2019-04-23:
Slight refinement - exactly which ordering is required does depend on how the exception is going to be triggered. If it was an interrupt going to be enabled with `NVIC_EnableIRQ` or a device write, then `DMB` is sufficient, because we just need to order the two memory accesses to the vector table and the SCS/device; we don't need to hold up the instruction flow.

Other exception changes might need `DSB`- eg to ensure a write to SVCall vector completes before executing `SVC`. 

But I think DMB and DSB both stall for write buffer drain on M7 anyway, and `SetVector` is not likely to be in a fast path where any extra cost of DSB would matter.

closed 2019-05-09 by JonatanAntoni