ARM-software/CMSIS_5 #579: how to calculate the m and n of activation

pkuswg, 2019-04-27:
how to calculate the m and n of activation？I am confused about the following three ways：
1. calculate Qm.n from  w*x
2. calculate Qm.n from  w*x+b
3. calculate Qm.n from  relu(w*x+b)
Which way should I choose?
My purpose is getting the BIAS_LSHIFT and OUT_RSHIFT 。。。
thinks

majianjia, 2019-04-27:
Hi pkuswg,
Please check the answers for #3 in this comment: https://github.com/ARM-software/CMSIS_5/issues/541#issuecomment-468255380 for getting LSHIFT and RSHIFT. 

In your example, you might want to use the output of `2.  w*x+b` to calculate shifts. 
This is due to relu and pooling in CMSIS-NN are not capable to change the output Q format from their input. 
However, if you use sigmoid, or tanh, there will be a different story. 

The way you think like this probably due to you are using Tensorflow directly when the 'layers' are splited into a few operations, if i am correct. 
Fully connected or convolutions operations in tf are built separately, however, in CMSIS-NN are normally handled in one operation. 
Using which operation to calculate shift depends on what is available in CMSIS-NN. 
If you consider in the way like layers in Keras, these operations and calculations should be eaier to understand.

pkuswg, 2019-04-27:
> 您好pkuswg，
> 请在此评论中查看[＃3](https://github.com/ARM-software/CMSIS_5/issues/3)的答案：[＃541（评论）](https://github.com/ARM-software/CMSIS_5/issues/541#issuecomment-468255380)获取LSHIFT和RSHIFT。
> 此外，[NNoM的讨论](https://github.com/majianjia/nnom/pull/7#issuecomment-468775756)也更多地解释了它。
> 
> > 简而言之，它们应该通过以下公式计算：
> > output_shift = output_n  - （input_n + weight_n）
> > bias_shift =（input_n + weight_n） -  bias_n 
> > CMSIS-NN的output_shift固定为右移。该值需要反转output_shift =  -  output_shift
> 
> 在您的示例中，您可能希望使用输出`2. w*x+b`来计算班次。
> 这是由于CMSIS-NN中的relu和pool无法从输入中更改输出Q格式。
> 但是，如果你使用sigmoid或tanh，会有不同的故事。
> 
> 您认为这样的方式可能是由于您正在将“图层”拆分为几个操作时直接使用Tensorflow，如果我是正确的。
> tf中的完全连接或卷积操作是单独构建的，但是，CMSIS-NN通常在一次操作中处理。
> 使用哪种操作来计算班次取决于CMSIS-NN中可用的内容。
> 如果您像Keras中的层一样考虑这些操作和计算应该更容易理解。
> 
> 希望它
> 有所帮助，建佳

thanks, jianjia. you are right, i am using tensorflow, I will take your suggestion and try more times.

pkuswg, 2019-04-27:
thanks, jianjia. you are right, i am using tensorflow, I will take your suggestion and try more times.

closed 2019-06-18 by felix-johnny