ARM-software/CMSIS_5 #580: Make CMSIS DSP for Cortex-A9 (Renesas-RZA1x)

ua1arn, 2019-04-27:
Hello. Wee propose little path (author - XGudron GitHub member) for arm_math.h for make possibility of compile CMSIS DSP in part of Fourier transforms. Try to apply this. As next step, build script for Keil MDK can be provided. Today not have enough information about library archives naming conventions.

[cmsis-5.5.1-cortex-a9.zip](https://github.com/ARM-software/CMSIS_5/files/3124319/cmsis-5.5.1-cortex-a9.zip)

JonatanAntoni, 2019-04-29:
Hi @ua1arn,

Thanks for your contribution. Please bare in mind pull requests are typically easier to handle than zipped patch files.

Cheers,
Jonatan

ua1arn, 2019-04-29:
Can you add me in collaborators?

 

  _____  

(Renesas-RZ/A1x) (#580)

 

it on GitHub, or mute
2XL3ANCNFSM4HI467HQ>  the thread.
<https://github.com/notifications/beacon/AILAVX7AB6GPE4G3WWUWIWTPS2XL3ANCNFS
M4HI467HQ.gif>

JonatanAntoni, 2019-04-29:
Hi @ua1arn,

This project is open for contributions to everybody. You can create pull requests without any limitation.

Cheers,
Jonatan

ua1arn, 2019-04-29:
See
https://github.com/ARM-software/CMSIS_5/pull/582

christophe0606, 2019-05-29:
For reasons explained in https://github.com/ARM-software/CMSIS_5/pull/582#issuecomment-496790999 , we won't merge those changes.

> Wee propose little path (author - XGudron GitHub member) for arm_math.h to make possibility of compile CMSIS DSP in part of Fourier transforms.

Regarding possibility of customizing CMSIS-DSP and only include some parts of it (like only including the FFT tables you use) : it was recently introduced in CMSIS-DSP.

There are now lots of compilation flags to tune what you want to build and include.
There is also a cmake build which is making a bit easier to set those compilation flags. But it can be done without the cmake.

closed 2019-05-29 by christophe0606