ARM-software/CMSIS_5 #581: CMSIS compile error

zcy618, 2019-04-28:
hi dear arm:
Currently, I am using your CMSIS, but I meet one issue about it, it is the same as:
https://github.com/ARM-software/ML-KWS-for-MCU/issues/94

Currently, the workaround solution is that I fall back to commit: 1d777eeeba686ba8e1f3e96c1f9a9396ae17ac1f according to the comment:

navsuda commented on 26 Mar • 
@jeremyholleman It failed for me too, but works fine with an older commit (1d777eeeba686ba8e1f3e96c1f9a9396ae17ac1f) of CMSIS_5 where there is no TransformFunctions.c.

Could I know if it is one known issue please? Or could you help to check it please?
Thanks

majianjia, 2019-04-28:
I think it is due to the change of new DSP lib.

zcy618, 2019-04-29:
> I think it is due to the changes of new DSP lib. Please check the release notes to locate the problem

if you have some idea about it, could you help to check it please? 
Thanks

JonatanAntoni, 2019-04-29:
Hi @zcy618,

The referenced issue already reveals the problem. We introduced some "collective C files" which include all the other C files in a group. You must not include both, the collective and the individual files, in your project. 

It depends on your build system how to exclude certain C files. In a "CMSIS-Pack aware environment" (e.g. MDK, Arm DS, IAR EW) only the collective C files are added to the projects, automatically. Otherwise you need to manually take care about whats getting compiled. E.g. when importing CMSIS-DSP into an mbed project using mbed cli build system you need to remove all the collective C files.

Please let me know if that explanation helps to solve your issue.

Cheers,
Jonatan

JonatanAntoni, 2019-05-21:
Closing due to no further activity.

closed 2019-05-21 by JonatanAntoni