ARM-software/CMSIS_5 #583: SVD conversion into multiple header files

radhika-raghavendran, 2019-04-29:
I am in the process of generating a header file for a Cortex-M33 MCU with many peripherals. The SVD file is going to be quite huge and the resulting header file including all macros, enums and structs is expected to be quite huge.
My question is this: Is is possible to split the SVD into multiple header files? Can I generate multiple header files from one SVD? I would ideally like to have the structures, macros and enums related to one peripheral in a file.
The header for the MCU can in turn include all peripheral header files with the interrupt numbers and other information generated from the SVD file.

I have one  more question with respect to header file generation. In our Cortex-M33 MCU, the base address for secure and non-secure peripherals are quite different. For example, non-secure USART peripheral starts at **_0x41000000_** and the secure USART peripheral starts at **_0x41400000_**.  Is it possible to accommodate both secure and non-secure header register base addresses in the same _**baseAddress**_  tag for the perihperal?

Lastly, the MCU has multiple I2C perihperals. while generating the header file from SVD, the bitmap positions of only one I2C peripheral appears in the header file. The rest of the bitmap positions are missing. I read the SVD documentation and tried to use the fields there, but that does not help. I have attached the SVD as well as header file for your reference.
The SVD command I use is 
**_SVDConv.exe C:\Users\fg6zwb\Desktop\moab.svd -o C:\Users\fg6zwb\Desktop\ -b svd.log --generate=header --fields=struct --fields=macro --fields=enum --generate=partition_**

[moab.zip](https://github.com/ARM-software/CMSIS_5/files/3127140/moab.zip)

Please let me know if I need to create different issue requests for these three queries.

Thanks,
Radhika

JonatanAntoni, 2019-04-29:
Hi @radhika-raghavendran,

Unfortunately its currently not possible to generate multiple headers from one single SVD file.

The other questions are rather detailed and needs a closer look.

Cheers,
Jonatan

flit, 2019-04-29:
From a user perspective, it's far better to have a single header file. When working on the Freescale Kinetis SDK, we originally used separate header files for each peripheral (mostly because of historical reasons on my part). Customers *hated* this! So we quickly switched to a single header file with all register definitions. It's much easier to manage, anyway, since it's only one file versus a folder full of headers (implying an extra include path).

flit, 2019-04-29:
Regarding the secure/non-secure peripherals, is your device not using the Arm-recommended S/NS striping of memory attribution every 256 MB via the IDAU?

Anyway, you should treat S/NS views of a peripheral as separate peripherals. It's highly unlikely that code will need to access both views of a given peripheral instance.

radhika-raghavendran, 2019-04-30:
Both Secure and Non-Secure view of the peripheral have the same register definitions but different base addresses only. That is the reason I was hoping that **_SVDConv.exe_** can automatically add two views of the peripheral using an element tag which differentiates between S and NS view of the peripheral.
I was hoping for something like given below, in the SVD Schema definition, where the SVDConv executable can generate structure for a peripheral as _PeripheralName_S_ and _PeripheralName_NS_

> ****
> **<xs:complexType name="securityType">**
> **< xs:sequence>**
> **<xs:element name="baseAddress" type="xs:integer"/>**
>  **< 0 - Indicates base address for NS perihperal; 1- Indicates base address for S perihperal>**
> **<xs:element name="secure" type="xs:boolean"/>**
> **<xs:element name="description" type="xs:string" minOccurs="0"/>**
> **</xs:sequence>**
> **</xs:complexType>**
>   
> <xs:complexType name="peripheralType">
>     <xs:sequence>
>       
>       <xs:group    ref="dimElementGroup" minOccurs="0"/>
>       
>       <xs:element name="name" type="dimableIdentifierType"/>
>       
>       <xs:element name="version" type="stringType" minOccurs="0"/>
>       
>       <xs:element name="description" type="stringType" minOccurs="0"/>
>       
>       <xs:element name="alternatePeripheral" type="dimableIdentifierType" minOccurs="0"/>
>       
>       <xs:element name="groupName" type="xs:Name" minOccurs="0"/>
>       
>       <xs:element name="prependToName" type="identifierType" minOccurs="0"/>
>       
>       <xs:element name="appendToName" type="identifierType" minOccurs="0"/>
>       
>       <xs:element name="headerStructName" type="dimableIdentifierType" minOccurs="0"/>
>       
>       <xs:element name="disableCondition" type="stringType" minOccurs="0"/>
>       
>       <xs:element name="baseAddress" type="scaledNonNegativeInteger"/>
>       
>       <xs:group ref="registerPropertiesGroup" minOccurs="0"/>
>       
>       <xs:element name="addressBlock" type="addressBlockType" minOccurs="0" maxOccurs="unbounded"/>
> 	  **< When no security type is included; the peripheral is assumed to be NS >
> 	  < There can be at max two instances of security type; one for secure register address and one for non-secure register address >
> 	  <xs:element name="security" type="securityType" minOccurs="0" maxOccurs="2"/>**
>       
>       <xs:element name="interrupt" type="interruptType" minOccurs="0" maxOccurs="unbounded"/>
>       
>       <xs:element name="registers" type="registersType" minOccurs="0" maxOccurs="1">
>       </xs:element>
>     </xs:sequence>
>     <xs:attribute name="derivedFrom" type="dimableIdentifierType" use="optional"/>
>  </xs:complexType>

jkrech, 2019-04-30:
Thanks for your feedback.
Instead you could already today make use of the "derivedFrom" attribute for the second peripheral instance and override the name, base_address, description and protection element on the peripheral level.

radhika-raghavendran, 2019-04-30:
The new SVDConv.exe does not generate bitmap mask for all "derivedFrom" peripherals. The previous versions of SVDConv used to generate the bitmap mask and position defines for all "derivedFrom" peripherals.
Can you please look into it?
Thank you.

jkrech, 2019-04-30:
Since you have specified the headerStructName to be 'I2C' my expectation was that instead of
`#define I2C0_CFG_SLAVE_EN_Pos              (0UL)                     /*!< SLAVE_EN (Bit 0) */`
a 
`#define I2C_CFG_SLAVE_EN_Pos              (0UL)                     /*!< SLAVE_EN (Bit 0) */`
would be generated. I cannot see why you want per instance macros if you are using the same struct for all instances. Would you agree or am I missing something?
Nevertheless I do agree that the current behavior of version 3.3.25 seems incorrect.

radhika-raghavendran, 2019-05-02:
I would expect a macro like #define I2C_CFG_SLAVE_EN_Pos (0UL) /*!< SLAVE_EN (Bit 0) */ since the peripheral header name is I2C. I agree that per peripheral instance macro is unnecessary and a wasteful overhead.

radhika-raghavendran, 2019-05-20:
Is there any update to SVDConv executable? We have multiple instances of several peripherals like I2C, SPI, USART etc and would like to use a macro with the header name and not theindividual instance of the peripheral. The current SVD converter is not able to generate a macro like this 
"#define I2C_CFG_SLAVE_EN_Pos (0UL) /*!< SLAVE_EN (Bit 0) */"

Instead it generates something like this
"#define I2C0_CFG_SLAVE_EN_Pos (0UL) /*!< SLAVE_EN (Bit 0) */"

Thanks
Radhika

JonatanAntoni, 2019-05-20:
Hi @radhika-raghavendran,

No, there is no fix available for that issue, yet. Sorry.

Cheers,
Jonatan

radhika-raghavendran, 2019-07-09:
Hi @JonatanAntoni 
Is there any fix for enumeration names, mask and position bit names for peripherals with multiple instances? Something like this, when we have multiple instances of I2C and the group name and header name is I2C and not I2C0, I2C1 etc.

Expected value:
"#define I2C_CFG_SLAVE_EN_Pos (0UL) /*!< SLAVE_EN (Bit 0) */"

Instead it generates something like this
"#define I2C0_CFG_SLAVE_EN_Pos (0UL) /*!< SLAVE_EN (Bit 0) */"

JonatanAntoni, 2019-07-09:
Hi @radhika-raghavendran,

This change in the behavior was introduced by a contrary requirement. Unfortunately with the current approach we cannot easily fulfill both. We need to carefully work out the different situations that could occur in an SVD file. Hence a proper solution will take some time.

Sorry for any inconvenience the current behavior might cause.

Cheers,
Jonatan