ARM-software/CMSIS_5 #584: all DSP floating po;int operations should support double precision (float64_t) operands

bigbrett, 2019-05-03:
It seems a bit silly that ARM advertises that fpv5 has double precision FPUs, but there is extremely limited support for float64_t in the ARM DSP library. If we get a chip with a double precision floating point unit, we probably want to use it. 

For example, all the matrix operation functions have f32 variants, but the only thing it appears you can do to an f64 matrix is invert it (unless I'm missing something in the source code....can't find any other f64 functions except `arm_mat_inverse_f64`). 

Is there any way we could f64's treated as first class citizens in the DSP library?

christophe0606, 2019-05-13:
@bigbrett f64 support is definitely in our plans but I cannot share any timeline.

udoe, 2019-06-11:
I do need support for float64_t as well, specifically for the FIR Interpolator filter.

JBecka, 2019-11-22:
Hello, 
our project is based on an ARM Cortex-M7 SoC with DP FPU.
We would like to perform FFT on 2048 points using double precision (f64).
At the moment the support of double precision floating point format in the CMSIS-DSP library seems to be very limited.
This core has been there for some time already, when we selected this device for our next project, we didn't expect that support of double precision may be missing in CMSIS-DSP.
Considering the number of similar comments appearing here, 
could you please treat this request with priority?
Thank you.

christophe0606, 2020-01-23:
There are no specific intrinsics for float64. So converting a function from float32 to float64 is basically just replacing the types. Then the compiler will manage the details.

As a consequence, since there are many other things to solve and improve in the CMSIS-DSP, we won't focus on this.

But pull requests with new functions are welcome :-)

If there is really a specific need and an unanticipated difficulty, then we will consider it on a case per case basis.

closed 2020-01-23 by christophe0606

bowen-hong, 2021-02-25:
Genius! You are genius, bro. :)