ARM-software/CMSIS_5 #585: why cmsis_os no define osPoolDelete ?

Cherish-Gww, 2019-05-09:
When I use cmsis_os.h, and I found there is no define a function named osPoolDelete, that means when your create a Pool it will exist all the time.

JonatanAntoni, 2019-05-10:
Hi @Cherish-Gww,

Please take not CMSIS-RTOS is deprecated and has been superseded by CMSIS-RTOS2.

Are you actually using `cmsis_os.h` (i.e. CMSIS-RTOS) to adopt another RTOS your own? Or are you using RTX4 (i.e. reference implementation for CMSIS-RTOS).

The latter does not use dynamic memory allocation for memory pools. Hence `osPoolDelete` has never been considered.

Cheers,
Jonatan

Cherish-Gww, 2019-05-14:
@JonatanAntoni 
Thank you, I use freertos, use CMSIS version 1.02. it is too old for me, and I want use RTOS2, is there have a cmsis_os2.c for freertos?

Cherish-Gww, 2019-05-14:
@JonatanAntoni 
Oh I found here
https://github.com/ARM-software/CMSIS-FreeRTOS
thank you very much

JonatanAntoni, 2019-05-14:
Hi @Cherish-Gww,

take a look to [CMSIS-FreeRTOS](https://github.com/ARM-software/CMSIS-FreeRTOS). It is a ready to use packaging of vanilla FreeRTOS with a full functional CMSIS-RTOS2 adoption layer, based on latest CMSIS-RTOS2 API version 2.3.1 and FreeRTOS 10.2.0.

Cheers,
Jonatan

Cherish-Gww, 2019-05-14:
@JonatanAntoni 
I found this in cmsis_freertos.txt
   - \b osMemoryPoolAlloc: \token{not implemented}
   - \b osMemoryPoolDelete: \token{not implemented}
   - \b osMemoryPoolFree: \token{not implemented}
   - \b osMemoryPoolGetBlockSize: \token{not implemented}
   - \b osMemoryPoolGetCapacity: \token{not implemented}
   - \b osMemoryPoolGetCount: \token{not implemented}
   - \b osMemoryPoolGetName: \token{not implemented}
   - \b osMemoryPoolGetSpace: \token{not implemented}
   - \b osMemoryPoolNew: \token{not implemented}
and I no found osMemoryPoolDelete

JonatanAntoni, 2019-05-14:
Please move questions regarding CMSIS-FreeRTOS to the according project.

ilg-ul, 2019-05-14:
> take a look to CMSIS-FreeRTOS. It is a ready to use packaging of vanilla FreeRTOS with a full functional CMSIS-RTOS2 adoption layer, based on latest CMSIS-RTOS2 API version 2.3.1 and FreeRTOS 10.2.0.

this should be a retorical question, but just to be sure, can you confirm that this project passed the CMSIS RTOS validation suite?

JonatanAntoni, 2019-05-14:
Closing this because it does not belong to CMSIS. Please raise questions regarding CMSIS-FreeRTOS to the according project.

closed 2019-05-14 by JonatanAntoni