ARM-software/CMSIS_5 #587: NN Missing Functionality UpSample

quinnabrvau, 2019-05-10:
I was noticing that there isn't an upsampling function which would commonly be used in CNN's to return the shape of a signal back to the original shape after pooling operations. I put together what I would consider a reference function with some tests.
https://gist.github.com/quinnabrvau/7926b7594da5a5d4aafc8dd00b968740

majianjia, 2019-05-12:
Great, I would suggest you make a pull request to CMSIS sometime later. 
I met the same problem before and [here is my implementation](https://github.com/majianjia/nnom/blob/3976855d003a7344bbba02e1ca3c31f191202f31/src/nnom_local.c#L184). It was verified in 1D/2D data. 
I saw you are also looking for deploying keras mode using CMSIS-NN. 
Please do check the [NNoM framework](https://github.com/majianjia/nnom) which can deploy Keras model directly to MCU.

quinnabrvau, 2019-05-12:
Thanks, I wanted some other thoughts before I opened another pull request, the NNoM framework looks handy. I do want to better understand the quantization step and how it effects the algorithm so I will probably keep playing with the other library I'm working on.

majianjia, 2019-05-16:
Thanks,
I am looking for your implementation. If it is optimized using ARM Instruction will be even more grateful.