ARM-software/CMSIS_5 #594: Cmsis compiler errors undefined reference to `arm_convolve_HWC_q7_RGB'

ccu1tn, 2019-05-20:
Dear everyone

I trying to the compiler arm_nnexamples_cifar10.cpp by mbed. but it appears one error (the error in the picture)
This is the header file and **arm_convolve_HWC_q7_RGB** function defined in **arm_nnfunctions.h** file
[image]
This is error
[image]

How to I can fix it?
Many thanks

JonatanAntoni, 2019-05-20:
Hi @ccu1tn,

The error message tells me that `arm_convolve_HWC_q7_RGB` cannot be linked. The function you try to use is defined in `arm_convolve_HWC_q7_RGB.c`. Are you sure this source file is included for compilation into your mbed project? The output only shows a single source file that is compiled. I guess you need to fix your mbed project setup. In doubt I suggest to get in touch with the mbed team.

Cheers,
Jonatan

ccu1tn, 2019-05-20:
HI @JonatanAntoni ;
yes, the source file is included for compilation into my mbed project. This is PATH I have configed for mbed.
[image]
This is all file in my folder to compile 
[image]
 It is only an example, I want to compile convolution functions with mbed.

JonatanAntoni, 2019-05-20:
I cannot spot `arm_convolve_HWC_q7_RGB.c` in your screenshots. On the screenshots it looks like you have copied all the header files. What about all the sources?
As I stated earlier: To me this looks like a build system issue. From CMSIS point of view I can only tell you that you need to include at least `arm_convolve_HWC_q7_RGB.c` into your project.

ccu1tn, 2019-05-21:
Dear @JonatanAntoni 

I included all library *.c file in my project, But, It not work. 
This is an error. 
[image]

JonatanAntoni, 2019-05-21:
Closing because this is a build system issue in the first place. Please get in contact with the mbed community about the build system.

closed 2019-05-21 by JonatanAntoni

ccu1tn, 2019-05-21:
Many thank @JonatanAntoni