ARM-software/CMSIS_5 #598: CMSIS DSP biquad filter seems could not support multiple stages

rednoah91, 2019-05-23:
In https://github.com/ARM-software/CMSIS_5/blob/9e28408ac965c528f6771d479abad505833683fd/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_f32.c#L338

should be ``` b0 = pCoeffs[0];
    b1 = pCoeffs[1];
    b2 = pCoeffs[2];
    a1 = pCoeffs[3];
    a2 = pCoeffs[4]; ``` ? The coefficient in every stage should be fixed value?

In https://github.com/ARM-software/CMSIS_5/blob/9e28408ac965c528f6771d479abad505833683fd/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_f32.c#L473

tt looks the pStates value will not be used at next stage if the pointer not be reset.

christophe0606, 2019-05-23:
Biquad filter is supporting several stages.

pCoeffs is containing coefficients for all stages.

pState is containing states for all stages.

pState for a stage will be reused by the same stage only when you compute another block of data. 

Have a look at this [comment](https://github.com/ARM-software/CMSIS_5/issues/508#issuecomment-481663284) to see how to implement a Biquad filter with CMSIS-DSP.

The example in the comment is using 1 stage. But there are tests with several stages in the CMSIS-DSP tests and in the Python wrapper example.

rednoah91, 2019-05-24:
Got it~ thx @christophe0606

closed 2019-05-24 by rednoah91