ARM-software/CMSIS_5 #600: Suggestion to add bit-field macros for the FPSCR exception flags

ioannisg, 2019-05-28:
To the best of my knowledge, the CMSIS headers do not define macros to represent the FPU exception bits on the FPSCR, for ARMv7-M and ARMv8-M architectures with FP extension implemented.

As an example, it would be nice to have macros such as : `FPSCR_IOC_Msk` (Invalid Operation exception)

It would be nice to have these macros directly in CMSIS, so we won't need to implement them in the application code.

JonatanAntoni, 2019-05-28:
Hi @ioannisg,

Thanks for your feedback. We'll check that part.

Cheers,
Jonatan

JonatanAntoni, 2019-07-26:
Hi @ioannisg,

sorry for reporting back that late. Access to `FPSCR` is given in the compiler specific headers because its a special function register. The individual bit definition in turn is not compiler specific hence I'd not add the bit macros next to the access functions.

Probably adding the bit macros to the affected core header files (i.e. cm4, cm7, cm33, cm35p, sc300, armv8mml and armv81mml) would be the cleanest solution. I suspect all those share the very same bit definition for `FPSCR` but not necessarily.

Is it feasible for you to add those macros you have in mind and raise a PR? Perhaps next to `FPU_Type`?
We need to think about proper representation in Doxygen documentation as well.

Cheers,
Jonatan

RFC: @GuentherMartin

niedzwiecki-dawid, 2024-05-08:
Coming quite late here, but I agree having FPSCR bits definitions in CMSIS would be great. Any chances for that in near feature?

JonatanAntoni, 2024-05-08:
@niedzwiecki-dawid, feel free to contribute these as a PR to https://github.com/ARM-software/CMSIS_6