ARM-software/CMSIS_5 #603: padding calculation

anathema43, 2019-06-04:
I had questions regarding  padding .calculation of padding in pooling function

if my input dim =(29,45,8) (h,w,c) and output is(15,23,8) with stride =2 and kernel =2, what should be my padding value ?

JonatanAntoni, 2019-06-04:
Can you please more specific to which component your are referring to?
I suspect its CMSIS-NN, but its easier to assign your request to the right experts if you state this clearly.

anathema43, 2019-06-04:
Sorry. It is for CMSIS- NN.

majianjia, 2019-06-05:
Base on my understanding, this is what you need to calculate the padding. 

O = (W−F+2P)/S+1

> The possible values for the padding size, P, depends the input size W (following the notation of the blog), the filter size F and the stride S. We assume width and height are the same.

However, the most important point is this part  `(W−F+2P)/S` is normally ceiling. 
Vice versa, P is flooring. 

In your example, 
ceil(29 / 2) = 15, "padding" = 0
ceil(45 / 2) = 23, "padding" = 0

To investigate further, there are 4 different paddings instead of 1 number. They are left, right, up, and bottom. Which you can set differently if you do the padding manually.

In your example, if we need to ceil on width in you example (to make 29->30), we cannot add padding to right and left evenly because we can only add 1 column to the left or right. So it is now the backend's choice to add it to the left border or right border. 
The padding argument of Keras has presented this uncertainty https://keras.io/layers/convolutional/#conv2d

anathema43, 2019-06-05:
Thanks Majianjia.
I am still confused if such problem arises what to do.
but for now i changed the kernel size to 3 if input dimension is odd

majianjia, 2019-06-05:
> Thanks Majianjia.
> I am still confused if such problem arises what to do.
> but for now i changed the kernel size to 3 if input dimension is odd

You dont need to do anything, just floor the padding value from the above equation and pass to CMSIS-NN. In your example, pass 0 for both width and height.

Its nn lib's decision to choose to pad left or right. CMSIS-NN pads right same as Tensorflow if I am correct.
Its fine as long as they do the same.

anathema43, 2019-06-05:
# Maianjia,

I did that initallly but there was an accuracy drop more than i excepted in the microcontroller , then i was not sure if the drop was due to quantization or if i was giving wrong padding parameter. So to make sure i made the kernel 3 if input dimensions are odd and training the network again and hopefully when  i port to cmsis nn  it would get better results.

majianjia, 2019-06-05:
> # Maianjia,
>
> I did that initallly but there was an accuracy drop more than i excepted in the microcontroller , then i was not sure if the drop was due to quantization or if i was giving wrong [...]

From my experience, it probably not the padding's fault. I don't think a minor difference on the border will affect the result much unless it changes the shape of your output data. 

If quantization failed (especially overflowing), the result will be significantly worse however the model won't completely fail (such as 90% -> 60% not 90%->10%). If you see this symptom, it could be your quantization process's defect.

anathema43, 2019-06-05:
> From my experience, it probably not the padding's fault. I don't think a minor difference on the border will affect the result much unless it changes the shape of your output data.
>
> If quantization [...]

Yes, I will check the quantization  process next.
Thank you for the advice and clearing my doubts on the padding problem.
Your help is really appreciated.

felix-johnny, 2019-06-12:
@anathema43 , @majianjia .. There are some new documentation that you might find interesting.. https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/converting-a-neural-network-for-arm-cortex-m-with-cmsis-nn/single-page

closed 2019-06-12 by felix-johnny

majianjia, 2019-06-12:
@felix-johnny 
Thank you for mentioning the document.  
Very good tutorial I wish I had read it before