ARM-software/CMSIS_5 #609: CMSIS_5 on CORTEX M4 STM32F4xx

SylvainLaurent37, 2019-06-17:
Hi everyone.

I'm working on STM32 with Keil and i would to use CMSIS_5 and particularly DSP/FFT.
Today I'm developping with an old library.

Does anyone have a template project with Keil using CMSIS_5 ?
Thank you very much for your help and sharing.

ReinhardKeil, 2019-06-17:
What STM32 device are you using?

SylvainLaurent37, 2019-06-17:
I'm using an STM32-discovery board with STM32F429ZI.

ReinhardKeil, 2019-06-17:
When you use the examples that are part of the Keil STM32F4 pack, then the current version of the CMSIS pack is automatically used.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

SylvainLaurent37, 2019-06-17:
Thanks for your response.
I use an RLink debugger and I would convert µVision project to Ride7 project.
When I try to build the project : "cc1.exe: fatal error: armv7e-m/fpu: No such file or directory"

ReinhardKeil, 2019-06-18:
Well the error message is not descriptive enough to understand the problem (which is most likely in the import step).

Did you check that the project builds under MDK?

ReinhardKeil, 2019-06-24:
Closing as there is no further follow up.

closed 2019-06-24 by ReinhardKeil