ARM-software/CMSIS_5 #610: Question to exclusive access with LDREX/STREX

joabeck, 2019-06-19:
Hello,
Because the documentation regarding LDREX and STREX is a little bit unclear to me, i have the following question:
Is it possible to use an RTOS-independent exclusive access using the instructions LDREX and STREX without interfering with the RTOS?

I like to use this exclusive access in different threads and in interrupt functions.

Thank you for your support.
Best regards
Joachim

JonatanAntoni, 2019-06-24:
Hello @joabeck,

You might want to use the atomic functions used in RTX5 as a reference about using `LDREX` and `STREX` instructions, e.g.
https://github.com/ARM-software/CMSIS_5/blob/01c7adb7685da540be9297b5a93e6640ea3333ce/CMSIS/RTOS2/RTX/Source/rtx_core_cm.h#L474-L489

Using you own exclusive access sequences should not interfere with the RTOS. The monitor is claimed by any `LDREX` instruction and cleared by the according `STREX` instruction, or by `CLREX` instruction, or when an exception occurs. It doesn't actually check whether a certain memory area is accessed concurrently. It simply assumes a concurrent access might occur if the sequence is interrupted. In that case you need to repeat the sequence.

Cheers,
Jonatan

joabeck, 2019-06-25:
Hi Jonathan,
Thank you for your complete and fast answer!
Best regards
Joachim

closed 2019-06-25 by joabeck