ARM-software/CMSIS_5 #619: RTX5 component viewer not working in MDK 5.28

balamk07, 2019-06-29:
Hi,

We recently updated from MDK 5.26 to MDK 5.28, and RTX5 updated to CMSIS 5.5.1 from CMSIS 5.4.0. RTX5 component viewer used to work with 5.26 doesn't work with the updated MDK. 

The viewer seems to initialize properly but fails as soon as osKernelStart() is called. Issue could be related to https://github.com/ARM-software/CMSIS_5/issues/522

[image]

We are relying on this tool for system tuning/debugging and would highly appreciate a fix at the earliest.

Thanks,
Bala
Analog Devices Inc.

VladimirUmek, 2019-07-01:
Hi Bala,

do you see any error message displayed (in the Command window)?
In case if there is no error message, can you describe how is the first thread started in your application?
And for example, if your application is complex, did you try to create a simple application with just one thread?

Regards,
Vladimir

balamk07, 2019-07-01:
Hi Vladimir,

Error message gets displayed in RTX Component viewer window as soon as  osKernelStart() called. I created a simple application that creates just one thread and no other RTOS objects before osKernelStart(). Since raising the issue, i found couple of other posts related to this.

http://www.keil.com/support/docs/3952.htm
http://www.onarm.com/forum/63790/

Regards,
Bala

VladimirUmek, 2019-07-01:
It does not look like your issue is related to the E306: Execution problem based on the screenshot you provided above.
Could you please log in to support system at https://support.developer.arm.com/ and attach RTX5 based reproducer project? We can then debug the application, find the problem and make the necessary steps forward.

Thanks,
Vladimir

balamk07, 2019-07-02:
Hi Vladimir,

I found the source of the issue, it is due to thread attribute settings (i am using custom thread configuration) that was corrupting the RTX component viewer object. For some reason, MDK 5.6 and RTX kernel 5.4 didn't pickup this issue and was working fine. Even though MDK 5.8 / RTX kernel 5.5 doesn't return an error during thread creation but the component viewer fails as soon as kernel start is called. The issue has been resolved and can confirm that RTX Component viewer on MDK 5.8 works fine. Sorry for the trouble and thank you very much for your support. 

The issue can be closed now.

Regards,
Bala

closed 2019-07-02 by balamk07

VladimirUmek, 2019-07-03:
Hi Bala,

if you describe what you are trying to achieve, we can take a look into RTX5 component view and maybe make it more robust, so that it can work for you in the future?