ARM-software/CMSIS_5 #622: arm_lms_norm_f32() documentation

Abhipray, 2019-07-06:
The documentation for arm_lms_norm functions says: 

> Filter coefficients are represented as fractional values and coefficients are restricted to lie in the range [-1 +1)

I don't see how in the code for arm_lms_norm_f32() for example, the coefficients are restricted to lie in the range [-1, +1). This came up because I'm observing learned weights that are greater than 1 which seems reasonable.  Can someone point me to what this line in the documentation is referring to?

christophe0606, 2019-07-08:
The comment in indeed wrong for a f32 implementation. It is also referring to a postShift so it is clearly a comment which applies only to fixed point implementations.

christophe0606, 2020-01-22:
The comment was updated as part of the big commit https://github.com/ARM-software/CMSIS_5/commit/814cf321e1463d1401105e215de2e72b88d93998

closed 2020-01-22 by christophe0606