ARM-software/CMSIS_5 #623: __SAUREGION_PRESENT not generated using SVDConv.exe

radhika-raghavendran, 2019-07-08:
Hi,

I am using SVDConv.exe version 3.3.25 for generating the partition file for a new Cortex-M33 MCU from our company.  I have included the description for SAU regions in SVD file and generate the partition header file using SVD conversion tool. When the element 
sauNumRegions = 8 is defined in SVD, the corresponding symbol in partition header is **#define __SAU_REGION_PRESENT                              1**  
The standard **TZ_SAU_Setup** function expects the macro to be **__SAUREGION_PRESENT** and not **__SAU_REGION_PRESENT**
SVD conversion routine generates the wrong preprocessor symbol for SAU region.
The sample ARMCM33_TZ.h file also defines the symbol to be **__SAUREGION_PRESENT** and not **__SAU_REGION_PRESENT**

JonatanAntoni, 2019-07-08:
Looks like a dup to #561. Should have been fixed in 3.3.26. Can you please double check if the latest version from develop branch works for you?

radhika-raghavendran, 2019-07-09:
I downloaded SVDConv from development branch and tried to run it on Windows 10 command prompt.
I get an error and am not able to run the program.Attached is a screenshot of the error.
[image]

thorstendb-ARM, 2019-07-09:
This seems to be true for all Windows / Linux binaries that are stored on Large File System. They load as html file (open as text).

radhika-raghavendran, 2019-07-09:
Got it! Downloaded as RAW file and generated the header.
Issue remains the same with this version also. Attached is a snapshot of the error. It should be generated with #define __SAUREGION_PRESENT 1 and **not** #define __SAU_REGION_PRESENT

radhika-raghavendran, 2019-07-09:
[image]

thorstendb-ARM, 2019-07-09:
SVDConv v3.3.27

fixed:
- define __SAUREGION_PRESENT (was: define __SAU_REGION_PRESENT)

see 74d8ed0

JonatanAntoni, 2019-07-09:
Thanks @radhika-raghavendran for spotting this issue. May I ask you to double check the fix provided asap? The latest version is subject for release.

radhika-raghavendran, 2019-07-09:
Yes, the fix is working as expected.
Thank you.

closed 2019-07-09 by radhika-raghavendran