ARM-software/CMSIS_5 #626: reference op doesn't implement saturating arithmetic.

zakk0610, 2019-07-11:
Hi,
fixed point functions are saturating arithmetic, but it looks like the some reference ops don't implement this behavior. (ex. ref_negate, ref_abs, ref_shift, ref_cmplx_conj..)

for example:
https://github.com/ARM-software/CMSIS_5/blob/d9a6b2e53a74acfbceee702f12eab5213d986016/CMSIS/DSP/DSP_Lib_TestSuite/RefLibs/src/BasicMathFunctions/abs.c#L29-L40

https://github.com/ARM-software/CMSIS_5/blob/d9a6b2e53a74acfbceee702f12eab5213d986016/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_q15.c#L112-L126

Is this a bug or am I missing out something?

Thanks.

christophe0606, 2019-07-11:
You are right that it is not correct. 

We don't like the old approach of comparing to a C "reference". In the future we would prefer to compare with reference patterns generated from SciPy or Matlab.

So, you can consider that the current DSP_Lib_TestSuite is deprecated. And in the future we plan on reworking the tests.

zakk0610, 2019-07-11:
@christophe0606 Got it. Thanks!

closed 2019-07-11 by zakk0610