ARM-software/CMSIS_5 #636: SVDConv does not pad bitfields

kjassmann-renesas, 2019-07-25:
Renesas MCUs have some registers that must be accessed with 32-bit access, but have no defined bitfields in the upper bits.  The structure for one of these registers as generated by SVDConv is here:

`  union {
    __IOM uint32_t SFMCMD;                      /*!< (@ 0x00000014) Communication Mode Control Register                        */
    
    struct {
      __IOM uint32_t DCOM       : 1;            /*!< [0..0] Selection of a mode of communication with the SPI bus              */
    } SFMCMD_b;
  } ;`

Recent versions of the GCC and AC6 compilers read and write to the DCOM bitfield of this register with ldrb and strb (when the peripheral requires this register to be accessed with ldr and str).

After some research, we found this:

> Note: Due to restrictions of the C/C++11 memory model, write accesses are not allowed to touch non bit-field members. It is therefore recommended to define all bits of the field's type as bit-field members.
Source: https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html

So we expect SVDConv to define all bitfield members required to reach the <size> defined in the SVD and generate something like:

`  union {
    __IOM uint32_t SFMCMD;                      /*!< (@ 0x00000014) Communication Mode Control Register                        */
    
    struct {
      __IOM uint32_t DCOM       : 1;            /*!< [0..0] Selection of a mode of communication with the SPI bus              */
      __IM  uint32_t            : 31;
    } SFMCMD_b;
  } ;`

I looked at the options in the latest SVDConv tool (v3.3.27), and I don't see any way to pad out the bitfield.  We are using the options --generate=header and --fields=struct.

Please let me know if there is any option we can define in the SVD or argument we can pass to SVDConv to ensure that all bitfields are fully defined.

Thank you,
Kristine

thorstendb-ARM, 2019-07-26:
Hi,
I added the padding generation to fill up bitfields. In my tests I did not see any side effects. Can you please test the attached version if this fits?

BR,
Th.

[SVDConv.zip](https://github.com/ARM-software/CMSIS_5/files/3434895/SVDConv.zip)

kjassmann-renesas, 2019-07-30:
Yes, this works for us.  Thank you!

kjassmann-renesas, 2020-02-05:
Do you plan to merge this back to the main development branch?

We are requesting another SVDConv update for C++, and we will need a version that supports both.

JonatanAntoni, 2020-02-06:
We are currently rethinking the way to distribute binary files. But basically the updated version of SVDConv will be part of the upcoming CMSIS release.

I guess this issues is solved with the preliminary version attached above. Please raise further requests in separate issues. 

Thanks,
Jonatan

closed 2020-03-26 by JonatanAntoni