ARM-software/CMSIS_5 #638: IAR project in NN example folder is out of date.

GorgonMeducer, 2019-07-29:
The IAR project in NN example folder seems out of date for the latest IAR 8.4x. 
Since different versions of IAR have different “understanding” about project files of other versions. It would be good to remove the IAR project until IAR fixes this annoying “FEATURE”.

TTornblom, 2019-07-29:
If the issue is that you get a warning about an unknown environment variable referring to an old CMSIS version on the first build, then this is just a warning and the project file will automatically be updated to the current version so that the warning will not be emitted on subsequent builds.
If you have another issue, then please elaborate.

GorgonMeducer, 2019-08-07:
The issue is solved. Thank you.

closed 2019-08-07 by GorgonMeducer