ARM-software/CMSIS_5 #639: osThreadFlagsSet returns 0

RePiTonE, 2019-07-29:
Hello,

I use CMSIS-RTOS2. I have 2 threads of same priority. I call osThreadFlagsSet from one thread and osThreadFlagsWait from the other, with option "clear flag". osThreadFlagsSet returns 0. I see with debug that thare is a peace of code that check if a thread is waiting on this flag and then because of the "clear flag" option clears the returned value instead of my flag value.

When checking the documentation I see that the return value should be the flag set. It is not mention any flag clear depending on option set to function osThreadFlagsWait.

Don't you think it could be usefull to return the flag set as said on the documentation instead of flag state after the osThreadFlagsWait processing ?
If the behavior is not planned to be updated, maybe the documentation should mention this behavior.
If osThreadFlagsSet was caled from interrupt the flag returned would be my flag instead of 0.

Thanks

JonatanAntoni, 2019-07-29:
Hi @RePiTonE,

CMSIS-RTOS2 is only API. I assume you are using RTX5 implementation?

I admit the documentation might be misleading in this case. In fact `osThreadFlagsSet` doesn't return the flags you actually set. It returns the flags stored just before returning. In your case another thread is already waiting on the flags you set. Hence the flags never get stored but are cleared immediately (due to the clear option).

In case calling `osThreadFlagsSet` from interrupt the order of processing is different. Hence the flag is actually stored and thus returned. You could provoke a similar behavior in thread context if you assure the other thread is not already waiting when the flags are set.

Does this makes sense to you?

Cheers,
Jonatan

RePiTonE, 2019-07-30:
Hello,

Thank you for your quick reply. Yes I use RTX5.
Yes I observed the behavior you described.

I was checking the flags returned to check if any error happened.
So can I consider that in case flag returned has not the highest bit set then there is no other possible error and my function was executed successfully?
Could the documentation be updated to reflect the real behavior ?

Thank you.

Best regards.
Pierre Peixoto
Engineer

[X]

________________________________
Cooper Securite S.A.S. ~ Siège social: Z.A. de la Graviere, Rue Beethoven, 63200, Riom, France ~ Lieu d'enregistrement au registre du commerce: ~ Numéro d'enregistrement: B334438827 ~ Numéro de TVA: FR00334438827

JonatanAntoni, 2019-07-30:
Yes, if the MSB is set (negative value if interpreted as signed) the value denotes an error code rather than a flag pattern. 
Any idea how we should add this specific detail to the documentation?

RePiTonE, 2019-07-30:
Maybe instead of :
"_It returns the flags set, or an error code if highest bit is set ..._"
Something like : 
"_It returns the new value of the flags, or an error code if highest bit is set ... If the function is NOT called from Interrupt Service Routine, flags set can be directly cleared if a thread was waiting on it_"

Maybe with a better English :-)

I'm not sure I can say "new value of the flag" as I don't know if the whole flags is returned or only the bit set by the call to _osThreadFlagsSet_.

Did you considered changing the behavior by returning the new value of flags even if a thread was waiting thoses flags ? Because technically, the flags are not cleared until the function _osThreadFlagsWait_ returns.

JonatanAntoni, 2019-07-30:
Actually, always "the new value of the flags" is returned. With the side-effect that between calling to and returning from `osThreadFlagsSet` not only new flags are set but set ones can be cleared by another already waiting thread. 

Technically `osThreadFlagsSet` simply returns the flags stored in thread control block just before returning, see [rtx_thread.c:1423-1442](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_thread.c#L1423-L1442).

I think your suggestion sounds reasonable to me. I'll check to enhance documentation that way.

Thank you very much,
Jonatan

KeilChris, 2019-09-26:
I have updated the documentation at commit bcc143b35. Closing.

closed 2019-09-26 by KeilChris