ARM-software/CMSIS_5 #642: Additional IRQ_MODE_xxx fields

ua1arn, 2019-08-01:
Renesas RZA1-family CPU r7s721020 (Cortex-A9) use lower bit if IRQ mode field (N-N or 1-N mode).
Manipulation method via IRQ_SetMode needs additional field definition.
irq_ctl.h:
```
// Encoding in some early GIC implementations
#define IRQ_MODE_MODEL_Pos          (13U)
#define IRQ_MODE_MODEL_Msk          (0x1UL << IRQ_MODE_MODEL_Pos)
#define IRQ_MODE_MODEL_NN			(0x0UL << IRQ_MODE_MODEL_Pos)	  ///< Corresponding interrupt is handled using the N-N model.
#define IRQ_MODE_MODEL_1N			(0x1UL << IRQ_MODE_MODEL_Pos)	  ///< Corresponding interrupt is handled using the 1-N model.

```

Modified files in attachment.

[Core_A.zip](https://github.com/ARM-software/CMSIS_5/files/3454410/Core_A.zip)

ua1arn, 2019-08-01:
Purge unneeded files from attachment.

ua1arn, 2019-08-01:
Example:
initial IRQs modes table (partial):
```
		/*  13 00000024 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_EDGE | IRQ_MODE_MODEL_NN),
		/*  14 00000024 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_EDGE | IRQ_MODE_MODEL_NN),
		/*  15 00000024 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_EDGE | IRQ_MODE_MODEL_NN),
		/*  16 00002020 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_1N),
		/*  17 00002020 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_1N),
		/*  18 00002020 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_1N),
		/*  19 00002020 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_0 | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_1N),
		/*  20 00000000 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_ALL | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_NN),
		/*  21 00000000 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_ALL | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_NN),
		/*  22 00000000 */	(IRQ_MODE_TYPE_IRQ | IRQ_MODE_DOMAIN_NONSECURE | IRQ_MODE_CPU_ALL | IRQ_MODE_TRIG_LEVEL | IRQ_MODE_MODEL_NN),

```

ua1arn, 2020-02-27:
Created https://github.com/ARM-software/CMSIS_5/pull/839

JonatanAntoni, 2021-06-08:
@ua1arn, The PR got closed without merge. Is this still required?

ua1arn, 2021-06-08:
Tray check... may be already done
--
Sent from Mail.ru app for Android Tuesday,