ARM-software/CMSIS_5 #647: numerical precision issues with arm_biquad_cascade_df2T_f32

bigbrett, 2019-08-05:
We are trying to use one of the chained biquad filters for a project, however we aren't seeing the numerical precision with the arm_biquad_cascade_df2T_f32() filter implementation that we would expect. To verify it wasn't any weird numerical instability, we implemented the filter ourselves (simple naive c implementation), and the results of our own implementation match perfectly with our MATLAB model of the filter response. 

We have a test input signal and a "ground truth" expected output based on a MATLAB model. 
When we use the CMSIS-DSP filter implementation, the difference between the filter output and the MATLAB ground-truth data is on the order of 10e-3, but when we wrote the filter implementation ourselves, it matches the MATLAB data exactly. 

Attached is a zip file containing two files: `double_notch_biquad_golden_truth_test_data_f32.h` and `CMSIS-DSP-bad-output.h`. 

The input data, coefficients used, and "truth" output of both the MATLAB and our corresponding simple C implementation is shown in double_notch_biquad_golden_truth_test_data_f32`. 

An array containing the erroneous result of the CMSIS DSP biquad function is shown in `DSP-bad-output.h`

We used the same interface functions and structures for our own naive implementation, and call both implementations through the same interface. 

Any idea why we are seeing a dramatic reduction in the accuracy with the CMSIS libraries? Is this to be expected?

[CMSIS-DSP-test-data.zip](https://github.com/ARM-software/CMSIS_5/files/3469454/CMSIS-DSP-test-data.zip)

Our own "naive" implementation of the filter, that we wrote to check ourselves, is below:

```
#pragma once

#include <arm_math.h>

typedef struct
{
    uint8_t numStages;
    float32_t* pState;
    const float32_t *pCoeffs;
} simple_cbq_instance;

typedef enum
{
    CBQ_B0,
    CBQ_B1,
    CBQ_B2,
    CBQ_A1,
    CBQ_A2,
    CBQ_COEFF_IDX_ENUM_SIZE,
} SimpleCbqCoeffIdx_t;

typedef enum
{
    CBQ_D1,
    CBQ_D2,
    CBQ_STATE_IDX_ENUM_SIZE,
} SimpleCbqStateIdx_t;

void simple_cbq_init_f32(simple_cbq_instance* inst, uint8_t num_stages, const float32_t* coeffs, float32_t* state)
{
    assert(NULL != inst);
    assert(NULL != coeffs);
    assert(NULL != state);
    assert(num_stages > 0);

    inst->numStages = num_stages;
    inst->pState = state;
    inst->pCoeffs = (float32_t*)coeffs;
}

void simple_cbq_update_f32(simple_cbq_instance* inst, const float32_t *src, float32_t* dst, uint32_t block_size)
{
    size_t blk=0;
    size_t stage=0;
    size_t coeff_base = 0;
    size_t state_base = 0;
    float32_t x = 0;
    float32_t y = 0;
    
    assert(NULL != inst);
    assert(NULL != src);
    assert(NULL != dst);
    assert(block_size > 0);

    for (blk=0; blk < block_size; blk++)
    {
        x = *src;
        y = 0;

        for (stage=0; stage < inst->numStages; stage++)
        {
            coeff_base = stage * CBQ_COEFF_IDX_ENUM_SIZE;
            state_base = stage * CBQ_STATE_IDX_ENUM_SIZE;
           
            // update filter output based on curent state
            y = inst->pCoeffs[coeff_base+CBQ_B0] * x + inst->pState[state_base+CBQ_D1];
            // update state
            inst->pState[state_base+CBQ_D1] = (inst->pCoeffs[coeff_base+CBQ_B1] * x) + (inst->pCoeffs[coeff_base+CBQ_A1] * y) 
                                              + inst->pState[state_base+CBQ_D2];
            inst->pState[state_base+CBQ_D2] = (inst->pCoeffs[coeff_base+CBQ_B2] * x) + (inst->pCoeffs[coeff_base+CBQ_A2] * y);

            x = y;
        }

        // apply the output, and walk input/output pointers to next element
        *dst = y;
        src++;
        dst++;
    }
}

```

christophe0606, 2019-08-06:
@bigbrett Thanks for the report. We are going to look at it.

christophe0606, 2019-08-06:
@bigbrett I have indeed been able to reproduce the "issue" and isolate the root cause.

If you look at the main loop for the unrolled version, you'see those lines:

```C
d1 = b1 * Xn1 + d2;
d1 += a1 * acc1;

d2 = b2 * Xn1;
d2 += a2 * acc1;
```

If you rewrite those lines into:

```C
d1 = b1 * Xn1 + a1 * acc1 + d2;
d2 = b2 * Xn1 + a2 * acc1;
```

you no more get the accuracy issue.

Here is the absolute error for both versions:
[image]

Can you confirm on your side that it is indeed the case ?

We intend on improving the tests for CMSIS-DSP using patterns generated by MATLAB or SciPy. So, this will be solved in a future version as soon as we have tests to track this issue.

bigbrett, 2019-08-06:
@christophe0606 interesting....so I was using the pack installer in Keil MDK to link against the DSP library. Since I couldn't see the source code (I'm sure I could find it somewhere on the install path on my computer but didn't want to edit that), I tried copying the files from the master branch on github directly into my project.

When I did this, the error issue went away. So whatevercode I was running via the pack installer had the precision issues, but the code on the master branch seems to work.

Any idea why this is the case? 

ARM MDK info:
```

IDE-Version:
µVision V5.25.3.0
Copyright (C) 2018 ARM Ltd and ARM Germany GmbH. All rights reserved.

License Information:
Brett Nicholas
Sarcos
LIC=KRMUI-Q4BSI-0Q1WS-4K83U-9KLFE-25W8M

Tool Version Numbers:
Toolchain:        MDK-ARM Plus: 4 user  Version: 5.25.2.0
Toolchain Path:    C:\Keil\ARM\ARMCC\Bin
C Compiler:         Armcc.exe        V5.06 update 6 (build 750)
Assembler:          Armasm.exe        V5.06 update 6 (build 750)
Linker/Locator:     ArmLink.exe        V5.06 update 6 (build 750)
Library Manager:    ArmAr.exe        V5.06 update 6 (build 750)
Hex Converter:      FromElf.exe        V5.06 update 6 (build 750)
CPU DLL:               SARMCM3.DLL          V5.25.2.0
Dialog DLL:         DCM.DLL              V1.17.1.0
Target DLL:             Segger\JL2CM3.dll      V2.99.29.0
Dialog DLL:         TCM.DLL              V1.35.1.0
```

Target: NXP LPC4330FET100

christophe0606, 2019-08-07:
@bigbrett I assume the difference on your side may be due to differences in the compilation : different compiler, compiler version or compiler flags. If the computation order is changed and instructions generated in a slightly different sequence then the rounding error will be different.

I don't think it is a compiler issue because on above MDK info you're using an AC5 compiler and I made my tests with an AC6 compiler. They are really different compilers.

bigbrett, 2019-08-07:
Sounds good. I will consider this issue "fixed" (sort-of) on my end, since the master branch code works. But glad I was able to key you in on an issue with a different compiler.

closed 2019-08-08 by christophe0606