ARM-software/CMSIS_5 #650: [NN S8] Need demo and scripts to know how to use it

parai, 2019-08-10:
Hi:
I find that there is s8 implementation which is for TF Lite micro.
The s8 is seems to be per-channel symmetric quantization, I am interested in it, and want to integrated it with my project [LWNN](https://github.com/lwnn/lwnn) to check how it can improve accuracy. I already find that Q16 can reach almost the same accuracy as float.
But in the offical project of tensorflow, I can't find anything related to s8(arm_convolve_s8), so I think it was under developping maybe. 
Can anyone of you from ARM share something about this?

felix-johnny, 2019-08-16:
@parai Thanks for the question and apologies for the delay. It is correct that there isn't a wrapper in TFLu to invoke arm_convolve_s8 yet. It is something that needs to be done and would like similar to this https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/experimental/micro/kernels/cmsis-nn
.

felix-johnny, 2019-08-16:
Also, for a demo, you can use the KWS demo(u8) in TFLu to try out what exists right now.

freddan80, 2019-08-16:
@parai There is a PR with instructions on how to integrate and test CMSIS-NN in TFLu. Check this: https://github.com/tensorflow/tensorflow/pull/30202

parai, 2019-08-16:
Hi @felix-johnny @freddan80:
Thanks for your information. For the s8, I somehow think it was implemented according to SYQ, below is the paper, but I am not sure:
http://phwl.org/papers/syq_cvpr18.pdf
As I see from this paper, that it needs training to get the output scale(output_mult) for each channel, is my understanding right?
If it was right, is it already supported by TF, pytorch or anything else, can you share how to train and convert it to run on CMSIS-NN with s8.
By the way, now I am developing an inference NN for MCU/Mobile/PC..., below picture show its architecture, CMSIS-NN is the backend for q8 and q16. I just want to try the s18 out, but I don't know how to generated the weights.
https://github.com/lwnn/lwnn
[image]

felix-johnny, 2019-08-21:
@parai not sure about the paper, but the CMSIS-NN s8 functions are implemented according to TFL's int8 quantization spec(https://www.tensorflow.org/lite/performance/quantization_spec) which has a mix of per-channel/axis and per-tensor quantization. Refer to the 'granularity' field in that.
As for getting an int8 model, you can do a post-training quantization. The links in https://medium.com/tensorflow/tensorflow-model-optimization-toolkit-post-training-integer-quantization-b4964a1ea9ba have information on post-training quantization. 
Not all int8 operators are currently supported by TFL micro and it isn't possible to run any int8 model of significance from TFLu(for now), but we see regular updates there and hopefully it should be possible soon. 
Ok, so I take it the CMSIS-NN back end is when using Cortex-M cores. I would suggest to try out the post training quantization to generate TFL micro int8 models(weights inclusive) and when there is more support added for int8 in TFLu, you can use the TFLu's interpreter to execute the models and use CMSIS-NN API's.

parai, 2019-08-22:
Hi @felix-johnny :
Thanks for your information, it's very valuable, I will look into it and give it a try.

parai, 2019-08-26:
Hi @felix-johnny :
I think that maybe there is some bug in NN s8.
According to https://www.tensorflow.org/lite/performance/quantization_spec, the bias should be with type int32_t, but in the NN code is int8_t, this is not reasonable.
Another thing is about arm_nn_sat_doubling_high_mult, the base is (1<<31), I think this is not good, that means the scale can only be in range 0-1, when I do intergration with my LWNN project, I change it the base to (1<<16) so that I can have a scale in big range.
Below commit is the thing what I do when I integrate with lwnn.
https://github.com/autoas/CMSIS_5/commit/074621132a1aaef3bb6ff670f2ba72edd36874b2
And here is the python code to generate the weights for s8 conv/dwconv:
https://github.com/lwnn/lwnn/blob/master/tools/lwnn/qformat.py LWNNQSFormatC
And another thing, is about arm_depthwise_conv_s8, I think there is bug, I can't make it works, you can check the commit above, the one I wrote according to q8 dwconv do works .

felix-johnny, 2019-08-27:
@parai .. you are right about the range of bias. I'll fix that. I'll get back with more details for arm_nn_sat_doubling_high_mult(). As for depthwise conv, what is the format of the filters that you used? The format is not the same as for conv.

parai, 2019-08-28:
Hi @felix-johnny :
The format is MxHxWxC, and M is generally 1.

felix-johnny, 2019-08-30:
@parai The int8 for bias is now fixed for conv and depthwise conv. Found a bug in the C implementation for depthwise conv and that is fixed now too.

parai, 2019-08-30:
@felix-johnny :
Thanks for your help, now with lwnn, I can run mnist with 98% accuracy for q8/q16 and s8.
https://travis-ci.org/lwnn/lwnn/builds/578827350

And by the way, the bias for arm_fully_connected_s8 and any other should all be int32.

felix-johnny, 2019-08-30:
@parai Good to hear! absolutely, it is in the plan and will be fixed next week. If you are using a M core with DSP extension(e.g, M4 or M7), you can try out the optimized versions.

I'll close this ticket now. Feel free to post for anything else..

closed 2019-09-02 by felix-johnny