ARM-software/CMSIS_5 #655: Gives -Wsign-conversion error

spe-ciellt, 2019-08-15:
The functions `__NVIC_SetVector()` and `__NVIC_GetVector()` in `CMSIS/Core/include/core_cm3.h` generates warnings when `-Wsign-conversion` flag is added to GCC 7-2018-q3-update.

I think these functions need an overhaul in general. Lots of strange casts back and forth. Therefore an issue and not a pull request, I don't dare to change anything there.

GuentherMartin, 2019-08-19:
The code could be changed to a better understandable code:

```
__STATIC_INLINE void __NVIC_SetVector(IRQn_Type IRQn, uint32_t vector)
{
  uint32_t *vectors = (uint32_t *)SCB->VTOR;
  vectors[(int32_t)IRQn + NVIC_USER_IRQ_OFFSET] = vector;
}
```

```
__STATIC_INLINE uint32_t __NVIC_GetVector(IRQn_Type IRQn)
{
  uint32_t *vectors = (uint32_t *)SCB->VTOR;
  return vectors[(int32_t)IRQn + NVIC_USER_IRQ_OFFSET];
}
```

spe-ciellt, 2019-08-19:
Much more clear, I say "go for it"!

kjbracey-arm, 2019-08-20:
@GuentherMartin - that reworked cm0 version can go horribly wrong, because you are working on a null pointer (0 cast to a pointer type).

See an example of the failure here: https://github.com/ARMmbed/mbed-os/pull/10534

That was an example of a compiler deliberately compiling a construct like:

    uint32_t *vectors = (uint32_t *)0x0U;
    vectors[(int32_t)IRQn + NVIC_USER_IRQ_OFFSET] = vector;

as

    UDF // undefined instruction

The act of doing the arithmetic before converting to a pointer stops it being a null pointer, making the compiler be less shirty.

kjbracey-arm, 2019-08-20:
One of my suggestions on that PR, which they didn't go with, was to instead do:

    uint32_t *vectors = (uint32_t *) (NVIC_USER_IRQ_OFFSET * 4);
    vectors[(int32_t)IRQn] = vector;

That changes it from "undefined behaviour" (accessing a null pointer) into "implementation-defined behaviour" (accessing some non-0 integer cast to a pointer).

kjbracey-arm, 2019-08-20:
Here's a compiler explorer example of clang x86-64 compiling that to nothing:

https://godbolt.org/z/qUjliX

The ARM version seemed to put an actual undefined instruction in.

kjbracey-arm, 2019-08-20:
GCC will also fail, but only when it inlines setvector with a constant parameter for `IRQn`, apparently.

GuentherMartin, 2019-08-20:
Hello @kjbracey-arm, 
Thank you for pointing out.
It is my fault because I did not test every single change I made. :-(

It causes only problems when we do not have a VTOR register.

kjbracey-arm, 2019-08-20:
It's a pretty evil bit of compiler behaviour. I wouldn't mind so much if it at least output a warning about accessing a null pointer before ignoring the code...

I've only caught `clang` to doing this, and it never occurred to me before I actually saw it happening in that other PR.