ARM-software/CMSIS_5 #658: arm_rfft_fast_f32() does not work for window size 128

andyp1per, 2019-08-19:
There is a bad bug in arm_rfft_fast_f32() when your window size is 128. All other window sizes work fine. Below is my code, it appears to result in some kind of infinite loop or memory violation - impossible to say as the board constant reboots:

    const int WINDOW = 128;
    float maxValue;
    uint32_t testIndex;
    float* test_window = new float[WINDOW];
    float* output_test_window = new float[WINDOW];
    for(uint16_t i = 0; i < WINDOW; i++) {
        test_window[i]= sinf(2.0f * M_PI * frequency * i / _fft_sampling_rate_hz) * 2000.0f;
    }
    
    arm_rfft_fast_instance_f32 _fft_instance;

    arm_rfft_fast_init_f32(&_fft_instance, WINDOW);
    /* Process the data through the CFFT/CIFFT module */
    arm_rfft_fast_f32(&_fft_instance, test_window, output_test_window, 0);

    /* Process the data through the Complex Magnitude Module for
        calculating the magnitude at each bin */
    arm_cmplx_mag_f32(output_test_window, test_window, WINDOW/2);

    /* Calculates maxValue and returns corresponding BIN value */
    arm_max_f32(test_window, WINDOW/2, &maxValue, &testIndex);

andyp1per, 2019-08-19:
I am testing on an STM32F7xx using the gcc 6.x toolchain

GuentherMartin, 2019-08-19:
Hello @andyp1per,

As far as I can see does file ./DSP/Source/TransformFunctions/arm_rfft_fast_init_f32.c from CMSIS release  V5.6.0 contain a bug regarding fftLen 128.
The bug is already fixed in the current version of the file.

You can replace file arm_rfft_fast_init_f32.c with the current version and rebuild the GCC binaries.
This should resolve your problem.

Martin

andyp1per, 2019-08-19:
I see it @GuentherMartin - or adding:
    arm_rfft_128_fast_init_f32(&_fft_instance);
works too. Many thanks. How soon before there will be another official release?

GuentherMartin, 2019-08-20:
Sorry, I can not tell when the next CMSIS release is planed.

JonatanAntoni, 2019-09-02:
Hi @andyp1per,

the next official release is not yet scheduled. Sorry for any inconvenience.

Cheers,
Jonatan

andyp1per, 2019-09-24:
I am using individual functions now

closed 2019-09-24 by andyp1per