ARM-software/CMSIS_5 #666: git lfs pull does not work

Ruide, 2019-08-29:
I am using ubuntu 16.04 on win10 wsl 2. 

When i git clone with git lfs installed, i get the following error:

17:45:14.699924 trace git-lfs: fetch CMSIS/Utilities/Win32/PackChk.exe [ecdd0baa82588538b59e1b07594344b3a23d41e4546dd3492aa765d37a95d427]
17:45:14.700595 trace git-lfs: fetch CMSIS/Utilities/Win32/SVDConv.exe [14f40abd8803a863b7447c957643260f81f5968a2a7c0d32304d558cb2944f52]
17:45:15.085524 trace git-lfs: HTTP: 403
17:45:15.087452 trace git-lfs: HTTP: {"message":"This repository is over its data quota. Purchase more data packs to restore access.","documentation_url":"https://help.github.com/articles/purchasing-additional-storage-and-bandwidth-for-an-organization/"}
17:45:15.092545 trace git-lfs: api error: This repository is over its data quota. Purchase more data packs to restore access.

JonatanAntoni, 2019-09-02:
Hi Ruide,

sorry for that inconvenience. Its a known issue caused by traffic limitation at GitHub.
Please try it again, we extended the traffic quota.

Cheers,
Jonatan

Ruide, 2019-09-03:
> Hi Ruide,
> 
> sorry for that inconvenience. Its a known issue caused by traffic limitation at GitHub.
> Please try it again, we extended the traffic quota.
> 
> Cheers,
> Jonatan

Yep, it works now. Thanks for your help!

Ruide

closed 2019-09-03 by Ruide