ARM-software/CMSIS_5 #669: arm_sqrt_f32 file could not be found

MishalJasmine, 2019-09-02:
Hi,

I am currently working on CMSIS DSP. In the file arm_math.h, a function arm_sqrt_f32 is defined and it in turn uses a function sqrt. I am not able to find the file arm_sqrt_f32.h. Eclipse keeps throwing an error at the function sqrt. 

In function `arm_sqrt_f32':
C:\CMSIS\ARM\CMSIS\5.6.0\CMSIS\Include/arm_math.h:5870: undefined reference to `sqrtf'

Can I get some help on this?

christophe0606, 2019-09-02:
Hi,

Which compiler are you using ?

MishalJasmine, 2019-09-02:
Hi,

I am using Cross ARM GNU C compiler.

christophe0606, 2019-09-02:
Normally sqrtf should be defined in the gcc libraries.
Is your error a link error or a compile error ? If link error, check the libraries on the gcc command line. -lm for instance.

MishalJasmine, 2019-09-03:
It is a linker issue. I have added arm_cortexM4l_math and GCC library path. But I still get this error. I also see there are some linker issues in CMSIS_pack 5.6.0. Is it because of that? My target device is MAX32660.

christophe0606, 2019-09-03:
Are you using -lm to link ? I think the missing sqrtf should be in the standard math library provided with the compiler.

MishalJasmine, 2019-09-03:
I am sure about -lm. I am using Eclipse and I have added the paths and the standard math library.  

[image]

MishalJasmine, 2019-09-04:
Paths to the compiler are automatically added when the CMSIS packs are installed. But the GCC paths and the standard math library paths need to added manually.

christophe0606, 2019-09-04:
I am sorry I can't really help more. I am sure your issue is related to some linking options.

MishalJasmine, 2019-09-04:
oh..thats fine..thank you @christophe0606 :-)

GuentherMartin, 2019-09-04:
Hello MishalJasmine,
you must link the math library (libm). Add '-lm' to your linker stage.

MishalJasmine, 2019-09-04:
I added the math library using -lm. It looks something like this. 

make all 
Building target: Demo.elf
Invoking: Cross ARM GNU C Linker
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mlittle-endian -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -Wl,-Map,"Demo.map" -u _printf_float -L C:\Users\Mishal\CMSIS_5\CMSIS\DSP\Lib\GCC -lm libarm_cortexM4l_math.a

But despite of giving the path to the math library it gives this error.
arm-none-eabi-gcc.exe: error: libarm_cortexM4l_math.a: No such file or directory
make: *** [Demo.elf] Error 1

JonatanAntoni, 2019-09-04:
Hi @MishalJasmine,

Did you write that compiler/linker command line yourself? It looks a bit strange to me. You simply give a file `libarm_cortexM4l_math.a` as input. There are no other `.o` files as input and no `-o <output>` options given. 

The error message simply tells me that no file `libarm_cortexM4l_math.a` can be found in the current directory. You need to give the full path to the file or use `-larm_cortexM4l_math` instead. The latter will instruct the linker to lookup a file named `libarm_cortexM4l_math.a` in all the library search directories.

Cheers,
Jonatan

MishalJasmine, 2019-09-04:
I didn't write command line. It is automatically generated by Eclipse. I need to add the respective file names and the path in the project properties section and it generates the Makefile for the same.

JonatanAntoni, 2019-09-04:
Please consider the Eclipse user group for getting detailed help about project configuration.

MishalJasmine, 2019-09-04:
-L C:\Users\Mishal\AppData\Local\Arm\Packs\ARM\CMSIS\5.5.1\CMSIS\DSP\Lib\GCC\libarm_cortexM4l_math.a

I was get able to get rid of the error with this. 

Thank you

closed 2019-12-17 by christophe0606