ARM-software/CMSIS_5 #679: Can SVD conv does not generate the correct header when using alternateRegister element?

radhika-raghavendran, 2019-09-09:
Hi,
I have a requirement where a particular register of a peripheral can be interpreted in two different ways based on the operating mode of the peripheral. The bit fields of this register are different based on the operating mode.
I tried to use the <alternateRegister> element in SVD to define the same register with different bit fields. SVD converter executable generates an error in this case.
The SVD and header file and error log is attached to this issue. 

[SampleHeader.txt](https://github.com/ARM-software/CMSIS_5/files/3588982/SampleHeader.txt)
[ErrorLog.txt](https://github.com/ARM-software/CMSIS_5/files/3588983/ErrorLog.txt)
[SampleSVD.txt](https://github.com/ARM-software/CMSIS_5/files/3588984/SampleSVD.txt)

Thanks,
Radhika

JonatanAntoni, 2019-09-09:
Hi Radhika,

I guess there is a typo in your SVD file that causes this issue. I assume you want to define `SEQ_SAMP` and `SIM_SAMP` to overlap at the very same memory location.
According to the [documentation](https://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html#elem_register) this should be done like:
```
    <register>
          <name>SEQ_SAMP</name>
          <addressOffset>0x04</addressOffset>
    <register>
    <register>
          <name>SIM_SAMP</name>
          <alternateRegister>SEQ_SAMP</alternateRegister>
          <addressOffset>0x04</addressOffset>
    <register>
```
I.e. the `alternateRegister` of the second one must match the `name` of the first one.
Please let me know if this solves your problem.

Cheers,
Jonatan

radhika-raghavendran, 2019-09-09:
Thank you very much for the prompt response. It did solve the issue at hand.

Regards
Radhika

closed 2019-09-09 by radhika-raghavendran