ARM-software/CMSIS_5 #680: convolutions with q15_t activations but q7_t weights

antofara, 2019-09-11:
Is there any suggested way to perform convolutions having input and output as q15_t and weights as q7_t? (excluding of course converting weights to q15_t and then use `arm_convolve_HWC_q15_*`, which would result, for large weights vectors, in a huge memory size which cannot be tolerated in my application). Any plan to add this feature? 
I have seen that internally `arm_convolve_HWC_q7_basic_nonsquare` (the function I would like to adapt, given a cortex M4 as target) uses `arm_q7_to_q15_no_shift` to cast inputs into q15 and move them into a buffer. If I just replace this function with `arm_copy_q15` and just provide the input directly as q15, and then replace
`*pOut++ = (q7_t) __SSAT((sum >> out_shift), 8);`
with
`*pOut++ = (q15_t) __SSAT((sum >> out_shift), 16);`
do you see any potential risk?

christophe0606, 2019-09-11:
@antofara It looks ok. 

But you'll also have to change the API of arm_nn_mat_mult_kernel_q7_q15 since the bias is in q7 and now, I assume, you need a bias in q15 like the inputs and outputs.

The implementation can remain the same since internally the bias is casted to q31 in arm_nn_mat_mult_kernel_q7_q15.

I think the bias_shift and input_shift can use the same values.

antofara, 2019-09-11:
@christophe0606 thanks for the reply. Actually biases are ok if kept as q7_t. In other words I am trying to create the correspondent version of `arm_fully_connected_mat_q7_vec_q15` but for convolutions.

I am now wondering what to do to also adapt the 'fast' convolution. Since `arm_convolve_HWC_q7_fast` uses `arm_q7_to_q15_reordered_no_shift`, to do the same thing I should provide the activations directly in the "reordered" way, which is quite impractical. What about starting from `arm_convolve_HWC_q15_fast ` and adapt it to use q7_ weights, then instead of copying them into pBuffer, I use `arm_q7_to_q15_no_shift` (basically the opposite I do with the basic case).

If I understood well, in that case I can provide the activations as q15_t and in the "normal order", right?

closed 2019-09-16 by antofara