ARM-software/CMSIS_5 #683: After stm32 internal flash program,system enters osRtxErrorNotify function

Pupil2013, 2019-09-16:
I am using stm32f405,when save data to internal flash,after program the data to flash,rtx enter osRtxErrorNotify function,stay in the osRtxErrorStackUnderflow case.
Is the flash write operation affect the rtos running?

Pupil2013, 2019-09-16:
Before the error happens ,there are still about 10k global dynamic memory left.

JonatanAntoni, 2019-09-16:
Hi @Pupil2013,

`osRtxErrorStackUnderflow` means that the thread running has exceeded its allocated stack. Stack size in RTX5 is not adopted, dynamically. So the free global dynamic memory doesn't affect the stack. 

You can try to enable the stack usage watermarking in RTX5. Than increase the stack size of the faulting thread. Run your application and watch the watermark to discover the actual stack requirements.

Does that help?

Cheers,
Jonatan

Pupil2013, 2019-09-16:
Thanks Jonatan.
It realy help.I forget to change the thread stack size when debuging.

JonatanAntoni, 2019-09-16:
Please consider closing this issue if your question has been solved, finally.

closed 2019-09-17 by Pupil2013