ARM-software/CMSIS_5 #690: RTOS2 SVCall, SysTick and PendSV configuration wrong for little endian Cortex M7?

graeme-hattan, 2019-09-26:
Looking the code in `rtx_core_cm.h`, I see the following lines to configure the PendSV and SVCall priorities respectively:

    SCB->SHPR[10] = 0xFFU;
    ....
    SCB->SHPR[7] = (uint8_t)(0xFEU << n);

Also `os_systick.c` has the following line for the SYSTick priority:

    SCB->SHPR[11] = SYSTICK_IRQ_PRIORITY;

Looking at the [documentation](http://infocenter.arm.com/help/topic/com.arm.doc.dui0646c/DUI0646C_cortex_m7_dgug.pdf) for the SHPR registers (section 4.3.8), they are defined as three 32-bit registers. It seems to me that the indexes used are only valid for big endian. If the processor is little endian, all that happens in each case is that reserved bits of the register are set.

What is going on here? Is the documentation wrong? It seems strange that each one of these is really a one byte register, yet `SHPR` is declared as an array of 12 `uint8_t`. It seems like it should either be and array of 3 `uint32_t`, with bitwise operations being used to access individual priorities, or that an endian check should be done somewhere.

The `NVIC_SetPriority()` and `NVIC_GetPriority()` functions also have the same issue.

RobertRostohar, 2019-09-26:
SHPR registers are in the System Control Block (SCB) and accessed through the Private Peripheral Bus (PPB) which is always accessed as little-endian (regardless of endianness settings). See section 2.1.5 in the mentioned document.

SHPR registers are also byte accessible (see second paragraph in section 4.3.8). Therefore it is legal to define them as an array of 12 uint8_t which makes it convenient to access when changing a single priority (design choice).

So there is no issue with the mentioned code in rtx_core_cm.h which has been verified and works as expected.

graeme-hattan, 2019-09-26:
Ok, I see, I have got my byte access order confused here. The numbers are valid for little endian and little endian is all that is ever used.

closed 2019-09-26 by graeme-hattan