ARM-software/CMSIS_5 #692: CMSIS NN like library for cortex A53 (raspberry pi)?

anathema43, 2019-10-01:


anathema43, 2019-10-01:
Sorry Jonatan I again forgot the label.

felix-johnny, 2019-10-01:
@anathema43 For Cortex-A, it would be the ARM Compute Library(ACL) that is the counterpart for M core's CMSIS-NN. https://developer.arm.com/ip-products/processors/machine-learning/compute-library has some startup info and the link to GitHub as well. 

When you say arm compute, is it the aforementioned compute library that you are referring to? I'll assume that is the case and answer. Please clarify if it isn't what you meant. 

ARM NN is a framework which supports multiple ML  frameworks(e.g. TensorFlow, Caffe) to use an underlying target(e.g A class via ACL, mali GPU's via ACL or ARM ML Hardware) efficiently. That said, an application can directly use the ACL just like how you can directly call API's in CMSIS-NN if it wishes so.

You can checkout this blog for more info on ARM-NN.
https://community.arm.com/developer/tools-software/tools/b/tools-software-ides-blog/posts/arm-nn-the-easy-way-to-deploy-edge-ml?_ga=2.244756224.402296189.1569841401-1263479483.1552291412&_gac=1.158912840.1569935259.Cj0KCQjw8svsBRDqARIsAHKVyqF4MYoSsu4UTPkOlSLLEy2nYfSszjdUrz0yZA3KA0X7VIlyoHh6-oEaAlkREALw_wcB

anathema43, 2019-10-02:
@felix-johnny  thank you for your time.  yes i meant the arm compute library. I was checking the arm compute. I was hoping if you could suggest me where do I start from ARM NN SDK or just Compute Library. Would using ARM NN be easier than Compute Library?

felix-johnny, 2019-10-02:
@gmiodice .. could you pitch in with a reply here?

gmiodice, 2019-10-02:
Hi @anathema43 ,

and many thanks for your interested on ACL and ArmNN. I think the best place to start playing with ACL is our github repo (https://github.com/ARM-software/ComputeLibrary). The documentation provided should help you to compile for Cortex-A53 and other Arm processors like Cortex-A76 or Mali-G76 (GPU) to quote few. We also have few tutorials to understand better the library and below you can find the links:

**Develop your own network using the Arm Compute Library**
https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/running-alexnet-on-raspberry-pi-with-compute-library/develop-your-network-using-the-compute-library

**Run AlexNet on Raspberry Pi with Arm Compute Library**
https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/running-alexnet-on-raspberry-pi-with-compute-library

Also for ArmNN you can find helpful information on GitHub (https://github.com/ARM-software/armnn)
An interesting tutorial for ArmNN can be found at this link: https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/configuring-the-arm-nn-sdk-build-environment-for-tensorflow where you can configure the Arm NN SDK build environment for TensorFlow

Hope this can help. Of course if you need any further information, do not hesitate to write on GitHub ;)

Regards,
Gian Marco

anathema43, 2019-10-03:
@gmiodice  Thanks for the reply. I did run the alex net according to the blog and now i am looking at graph_vgg_16 example and trying to build my own network. IN this context if you could clear some of my doubts regarding creating the graph

1) most documents I read said to 1st initialize the tensor, configure the tensor( fix the operation i think) and then finally allocate. But in the example I dont see all this?  does graph method not require this? I did not find neon implementation of this model so i dont know if its not graph specific

2) in the vgg 16 example , the data before the dense layer is not flattened, does the library does this itself, because I read a issue where it said that I need to flatten.

3) pooling layer how do we define the  PadStrideInfo

4) Graph vgg 16 has in its output layer has a integer 5 what does it signify.

I know these are basic questions but i could not understand it. so reaching out for help.

thank you

gmiodice, 2019-10-04:
Hi @anathema43 ,

if you use the graph API, you do not need to call explicitly "init", "configure" and "allocate" because everything should be handled by the graph API itself. You need just to call layers that you need for your graph.

The data before the first dens layer (fully connected) is not flattened because the graph API automatically calls the flattening function (a.k.a. reshape) when it sees a conv layer followed by a fully connected layer.

For pooling you can pass the PadStride info into PoolingLayerInfo. It is the first argument required by pooling layer graph node

In the end, VGG16 has 5 in its output layer in order to return the first "most likely" 5 output classes. You can change this value up to the maximum number of classes available for your CNN

Thanks,

anathema43, 2019-10-04:
@gmiodice  Thanks you very much for your answers. Is there a documentation  where you can point me to so i can compile my graph. and not the whole library and examples again.
thanks a lot

closed 2019-10-21 by felix-johnny