ARM-software/CMSIS_5 #693: Migration from rtx4.5 to rtx5.x: osKernelSysTickMicroSec

hamgdandrea, 2019-10-07:
Hello,
with RTX CMSIS1 we were using the following API to get the time in microseconds from the kernel:

uint32_t tick = osKernelSysTick();
uint32_t time_us = osKernelSysTickMicroSec(tick);

These functions are not available anymore on CMSIS2 API.

Could you please explain how this can be achieved with CMSIS2?

JonatanAntoni, 2019-10-07:
Hi @hamgdandrea,

please find a comparison between CMSIS-RTOS API v1 and v2 at https://arm-software.github.io/CMSIS_5/RTOS2/html/os2MigrationFunctions.html.

The API v2 uses tick values, e.g. use `osKernelGetTickCount` to get the current OS tick. The tick frequency (in Hz) can be queried with `osKernelGetTickFreq`. Additionally you can query the sub-ticks with `osKernelGetSysTimerCount` and the sys tick timer frequency with `osKernelGetSysTimerFreq`.

Does that answer your question?

Cheers,
Jonatan

hamgdandrea, 2019-10-08:
Hi Jonatan

Thanks you for the fast reply!
BR
Giovanni

Giovanni Dandrea

Software Engineer

Hamilton Medical AG
Via Crusch 8
7402 Bonaduz
Switzerland

Phone:

+41 58 610 27 32

Fax:

+41 58 610 00 20

Mail:

gdandrea@hamilton-medical.com

Web:

www.hamilton-medical.com

[cid:IntelligentVentilationNewsletteremailFooter2_507a3e2d-7335-444f-b2d1-902b9e4a5235.png]<https://www.hamilton-medical.com/News/Newsletter-sign-up.html>

This message is confidential. It may also be privileged or otherwise protected by work product immunity or other legal rules. If you have received it by mistake, please let us know by e-mail reply and delete it from your system; you may not copy this message or disclose its contents to anyone. The integrity and security of this message cannot be guaranteed on the Internet.

closed 2019-11-06 by JonatanAntoni