ARM-software/CMSIS_5 #695: Array indexing bug in arm_bilinear_interp_f32()

FalkMusick, 2019-10-09:
File Path: CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_math.h.

For the correct output, the input values range starts at 1 and goes up to the respective matrix dimension according to the mathematical matrix indexing convention. A 2x2 Matrix may have the elements a11, a12, a21, a22. Valid input is then a float between 1 and 2 for the interpolation.

The IF guard however, assumes the input index is 0 to size-1. This will cause access at negative array indices when the input is 0. 

__STATIC_FORCEINLINE float32_t arm_bilinear_interp_f32(
  const arm_bilinear_interp_instance_f32 * S,
  float32_t X,
  float32_t Y)
  {
    float32_t out;
    float32_t f00, f01, f10, f11;
    float32_t *pData = S->pData;
    int32_t xIndex, yIndex, index;
    float32_t xdiff, ydiff;
    float32_t b1, b2, b3, b4;

    xIndex = (int32_t) X;
    yIndex = (int32_t) Y;

    /* Care taken for table outside boundary */
    /* Returns zero output when values are outside table boundary */
    if (xIndex < 0 || xIndex > (S->numRows - 1) || yIndex < 0 || yIndex > (S->numCols - 1))
    {
      return (0);
    }

    /* Calculation of index for two nearest points in X-direction */
    index = (xIndex - 1) + (yIndex - 1) * S->numCols;

    /* Read two nearest points in X-direction */
    f00 = pData[index];
    f01 = pData[index + 1];
[...]

christophe0606, 2019-10-10:
@FalkMusick Thanks for reporting this bug. Indeed, the index computation is not coherent with the IF guard and with the comments.

christophe0606, 2020-01-22:
It should be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/a8a19109541f6a147c64ea557ad0f5f51fe08f24

Now indexes for f32 versions should start at 0 like with other types (q31, q15 and q7).

closed 2020-01-23 by christophe0606