ARM-software/CMSIS_5 #696: Multiple definition in DSP package

uint32tMnstr, 2019-10-10:
Some mistake like "  multiple definition of `xxx' " while using DSP lib may exist.

Case:
tool-chain: arm-none-eabi
complie .a file success but mistake occur when using this file.

Reason:
In each DSP source directories, there is a ".c" file named the same as its directory name, which include every **cfile** from its path.
Thus including DSP lib will complie those cfile **twice** and link to the result.a if you just scan each directories and include all cfile in Makefile.
Anyway, it cames out successful to generate the .a file.
But it confuse users when using it.

christophe0606, 2019-10-10:
@uint32tMnstr I agree that it is confusing to have those files which are including all other C file of same folder.

It was introduced to simplify the build on some build systems.

I don't know if they can be removed for now but it is something to consider for the future.

Thanks for reporting it.

closed 2019-12-17 by christophe0606

igoryashkka, 2024-08-12:
if i will comment all files with .c it will work?