ARM-software/CMSIS_5 #704: How the input pre-processing was done with the code in arm_nnexamples_cifar10.cpp?

qwb1111, 2019-10-14:
/* input pre-processing */
  int mean_data[3] = INPUT_MEAN_SHIFT;
  unsigned int scale_data[3] = INPUT_RIGHT_SHIFT;
  for (int i=0;i<32*32*3; i+=3) {
		

    img_buffer2[i] =   (q7_t)__SSAT( ((((int)image_data[i]   - mean_data[0])<<7) + (0x1<<(scale_data[0]-1)))
                             >> scale_data[0], 8);
		
    img_buffer2[i+1] = (q7_t)__SSAT( ((((int)image_data[i+1] - mean_data[1])<<7) + (0x1<<(scale_data[1]-1)))
                             >> scale_data[1], 8);
		
    img_buffer2[i+2] = (q7_t)__SSAT( ((((int)image_data[i+2] - mean_data[2])<<7) + (0x1<<(scale_data[2]-1)))
                             >> scale_data[2], 8);

  }

the input pre-processing was done with the code upside.I did not understand it,how can we define INPUT_MEAN_SHIFT and INPUT_RIGHT_SHIFT?

qwb1111, 2020-06-11:
@felix-johnny Thanks a lot! After watching the  tutorial, I managed to use my own code to do the pre-processing and get the same consequence.So I think I am fully understanding how this part work now.

felix-johnny, 2020-06-11:
@qwb1111 That is great to hear :-) Thanks for using CMSIS.

closed 2020-06-11 by felix-johnny