ARM-software/CMSIS_5 #705: When using osMutexPrioInherit, priority is not restored after a timeout

graeme-hattan, 2019-10-14:
Recently we had a issue with a low priority thread locking up in a loop. It had locked a mutex with `osMutexPrioInherit` and, when a higher priority thread tried to lock the same mutex, this thread would lock up too.

The higher priority thread would timeout and go to the `READY` state, but it would never return from `osMutexAcquire`.  Looking at the lower priority thread, its priority was bumped to that of the higher priority thread at the start of the timeout as expected, but it continued to have the same priority after the timeout expired. Since it would keep running without releasing the mutex or doing anything that would cause it block, the higher priority thread remained in the `READY` state without running.

Looking at the code, I don't see anthing that will change the priority of the mutex owner when after `osMutexAcquire` times out. Presumably the priority should be set to that of the next highest priority thread waiting or `priority_base` around the points where `EvrRtxMutexAcquireTimeout` is called.

Note also that we have the round robin enabled and I expected it to kick in here, but since we have a thread of higher priority than the two mentioned that runs with a 2 ms delay, this doesn't happen. This is covered by issue #228.

JonatanAntoni, 2019-10-14:
Hi @graeme-hattan,

Thanks for pointing out this issue. We need to carefully analyze whats behind. I appreciate this sounds weird on the first glance. 

Cheers,
Jonatan

RobertRostohar, 2019-10-16:
Inherited thread priority needs to be restored also when osMutexAcquire (which caused the priority inversion) timeouts. This situation has been overlooked. We’ll look into fixing this.

RobertRostohar, 2019-10-18:
Thread priority is now restored also on mutex acquire timeout and should fix the reported problem.

closed 2020-03-26 by JonatanAntoni