ARM-software/CMSIS_5 #706: osThreadSetPriority can lower priorities temporarily increased by osMutexPrioInherit

graeme-hattan, 2019-10-14:
While investigating issue #705, I also noticed that if `osThreadSetPriority` is called on the owner thread of a mutex using `osMutexPrioInherit`, the temporary priority change will be overwritten.

Looking at the code for `svcRtxThreadSetPriority`:

```
  if (thread->priority   != (int8_t)priority) {
    thread->priority      = (int8_t)priority;
    thread->priority_base = (int8_t)priority;
    ...
```

To avoid this, only `priority_base` would be changed if the thread is an owner of a mutex that has a thread of greater or equal priority waiting. Presumably this would be fairly easy to fix if the mutex code set a flag to indicate this condition...?

JonatanAntoni, 2019-10-14:
Hi @graeme-hattan,

Thanks for drilling this down into the details. We'll check if that can be fixed.

Cheers,
Jonatan

RobertRostohar, 2019-10-16:
`osThreadSetPriority` is an explicit request by the user to set the thread priority. It should update the priority of the thread as requested regardless of what the previous priority was. This is the intentional design.

The temporary priority implicitly changed by the priority inversion is no exception and will be overridden by this explicit request.

The user needs to know what he is doing when priority inversion is being used mixed with explicitly setting priorities.