ARM-software/CMSIS_5 #715: SVDConv does not provide uint64_t

vitkorob, 2019-10-24:
When I use &lt;size>64&lt;/size> in &lt;register> I have strange error in stdout.

```
CMSIS-SVD SVD Consistency Checker / Header File Generator V3.3.27
Copyright (C) 2010 - 2019 ARM Ltd and ARM Germany GmbH. All rights reserved.

Arguments: "C:\projects\hammer\examples\uint64_t_DATE.svd" -o "C:\projects\hammer\examples" --generate=header --fields=struct --fields=macro --fields=enum -x "M216" -x "M347" -x "M317" -x "M318" -x "M356" -x "M209"

*** ERROR M219: C:\projects\hammer\examples\uint64_t_DATE.svd
  C Code generation error: Struct end-padding calculation error!

Found 1 Error(s) and 0 Warning(s).
```

When I change to &lt;size>32&lt;/size>, it works fine, but it does not correspond to the type &lt;dataType>uint64_t&lt;/dataType> that I want.

Examples (see register InfoRO->DATE):
 -  Normal SVD file- [uint32_t_DATE.txt](https://github.com/ARM-software/CMSIS_5/files/3769139/uint32_t_DATE.txt)
 -  With error - [uint64_t_DATE.txt](https://github.com/ARM-software/CMSIS_5/files/3769142/uint64_t_DATE.txt)

JonatanAntoni, 2019-10-25:
Thanks for raising that. We will take a look into your reproducers.

vitkorob, 2019-11-13:
Hello, Jonatan. Do you need any help with review?

JonatanAntoni, 2019-11-13:
Hi @vitkorob,

I discussed this with the developers already. We need to take a closer look, which will take some more time. Sorry for any inconvenience.

Cheers,
Jonatan

vitkorob, 2020-03-25:
Hi @JonatanAntoni 

What about plans to fix this issue?

JonatanAntoni, 2020-03-25:
No news, sorry.